FreeCAD/FreeCAD #11239: [Problem] clean_icon is giving bad filenames to Windows

jeffallen, 2023-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Seen during a full update of the addon metadata:

```
12:27:15  Traceback (most recent call last):
12:27:15    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_workers.py", line 916, in run
12:27:15      self.macro.fill_details_from_wiki(url)
12:27:15    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 315, in fill_details_from_wiki
12:27:15      self.clean_icon()
12:27:15    File "C:\Program Files\FreeCAD 0.20\Mod\AddonManager\addonmanager_macro.py", line 341, in clean_icon
12:27:15      with open(constructed_name, "wb") as f:
12:27:15  OSError: [Errno 22] Invalid argument: 'C:\\Users\\jalle\\AppData\\Local\\Temp\\FreeCAD\\Cache\\AddonManager\\MacroIcons\\120px-FCBmpImportScreenshot.png?20210923163201'
```

Having a ? in the filename is making Windows mad.

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.14
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-10-31:
Obsolete version. Closing unless you can reproduce on v0.21.1 (then please re-open)

closed 2023-10-31 by luzpaz

jeffallen, 2023-10-31:
Yeah, sorry, probably was fixed in f1ffaf3ba9a1cdf408334383e0ba5c6767919b2c.