FreeCAD/FreeCAD #11243: [Problem] Camera navigation out of control

maxwxyz, 2023-10-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the current weekly build the default navigation gets sometimes out of control. Normally I use the CAD navigation style around the center. The red sphere should indicate my center of rotation in the viewport and therefore be steady. 
In the following clip this is also the correct behavior in the beginning but sometimes the model jumps around and you can clearly see, that the center of rotation itself is rotating around what I guess is the scene/world center:

https://github.com/FreeCAD/FreeCAD/assets/6246609/29d486b7-f0be-423c-a618-c699884fb02e

The whole clip I am holding the middle and right mouse button.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34810 (Git)
Build type: Release
Branch: main
Hash: 196cbecde66a1c3a1b56b8c532ad1faedff4818b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Rexbas, 2023-11-01:
I cannot reproduce this, can you maybe share your file? And started this problem happening in 0.22.0dev.34810 or can you also reproduce this problem in the previous weekly build?

maxwxyz, 2023-11-01:
I just created a fresh file, only a box:
[sample2.zip](https://github.com/FreeCAD/FreeCAD/files/13226138/sample2.zip)

Same behavior:

https://github.com/FreeCAD/FreeCAD/assets/6246609/7ec2c4f5-a75b-4f83-b7b1-23fe952ed8d6

maxwxyz, 2023-11-01:
I have narrowed it down. Turning off animations in navigation style helps but I think the key is, that when zooming while orbiting the center of rotation gets changed. With CAD navigation you pan with middle mouse button. Hold down right mouse as well and you orbit. Releasing the RMB and keep holding the MMB and clicking RMB you zoom. When you click RMB for orbit again while holding MMB it glitches and sets a random center of rotation. I was not aware that I previously let go of RMB and clicked it again, so not sure if that was the cause.
Edit: it also happens when panning with MMB and then start rotating (keep MMB and press and hold RMB too) as seen in the beginning of the video:

https://github.com/FreeCAD/FreeCAD/assets/6246609/88226df1-09a5-4bf8-967f-eb7a615a3f65

Rexbas, 2023-11-01:
Haha, I was just about to ask if you were using rotation mode = window center. I can now reproduce it. I will see if I can fix it sometime this week.

luzpaz, 2023-11-01:
@maxwxyz thanks so much for the in depth testing and diagnosis!

Rexbas, 2023-11-01:
It also happens to other navigation styles and only with orthographic view, perspective view works fine for me.

maxwxyz, 2023-11-01:
Is this something which was introduced lately or is this also in the stable version? I am not sure because the rotation indicator is not available in 0.21 but I think it glitches sometimes too.

Rexbas, 2023-11-01:
I did a quick comparison between 0.21.1 and 0.22.0dev, it looks like they behave different so its probably a new problem.

maxwxyz, 2023-11-01:
I can reproduce in stable 0.21.
First pan with MMB and then add RMB for rotation and its bonkers (the animation which seem to start randomly is also very annoying 😄):

https://github.com/FreeCAD/FreeCAD/assets/6246609/6abe8a48-50f5-4475-8d0c-9ce5d63285d9

maxwxyz, 2023-11-02:
@Rexbas There is also this bug in the weekly build: when you set the rotation center, then zoom and then rotate (all operations after each other with releasing the mouse buttons), the rotation center changes and the red indicator orbits around as well. This is not the case in the stable 0.21.1

maxwxyz, 2023-11-24:
@Rexbas it is better but sometimes it still glitches and the center of rotation orbits around as well. Should I open a new issue or can we reopen this one?
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35176 (Git)
Build type: Release
Branch: main
Hash: dcd533af4bf0cb3dbc50c7d0e2fb579cad168819
Python 3.10.13, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.8
  * CurvedShapes 1.0.5
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.70
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.12
  * woodworking 0.21.30486
```

maxwxyz, 2023-11-24:
@adrianinsaval can you reopen please?

@Rexbas here a quick demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/223815d3-5884-4edc-ad39-fd5af291f8bb

Rexbas, 2023-11-24:
Thanks, I will take another look at it.

PR #11279 (2023-11-20): Gui: Fix window center rotation mode and orthographic view clipping
This PR fixes #11243 and re-fixes #6836. #9699, The original fix for #6836 was a huge improvement but I was still able to reproduce the cut corners and sometimes random glitches. With this fix that does not happen anymore, at least during my testing.

The new approach finds a bounding sphere of the scene, then it is repositioning the camera such that the `nearDistance` can be 0 and the `farDistance` the diameter of the bounding sphere. Finally, to fix #11243, we change the focal distance so the focal point stays the same.

Cur [... cut]
PR #11519 (2023-11-27): Gui: Update rotation center indicator after zoom
#11243 was not completely fixed. This PR fixes #11243. The problem also occurred when dragging and using the scroll wheel to zoom at the same time. This only happend with rotation mode = window center and _zoom at cursor_ enabled. @maxwxyz can you confirm this works?

The problem is resolved by updating the position of the rotation center, to the new focal point, after zooming. ~~This PR also introduces a small regression which is a flickering rotation center after zooming using the scroll wheel. However, when you start dragging again without zo [... cut]
closed 2023-11-27 by chennes