FreeCAD/FreeCAD #11247: [Problem] Open recent menu shows deleted files

hasecilu, 2023-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was making some clean-up in the FreeCAD-library, a lot of files replaces by a few.

The thing is that in the `Start` WB all the files that I opened before and were deleted stopped appearing which is good.
[image]

but the list of file in `File>Open Recent` was not updated meaning that all deleted files are still there and that's not expected, I mean certainly were openend recently but since they were deleted should not appear anymore in the same way as previous image.

[image]

And the file gives an error

[image]

### Full version info

```shell
[code]
OS: Arch Linux (XFCE/openbox)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34822 (Git)
Build type: Unknown
Branch: main
Hash: e4d6c328fcc110bc274a31ef750d36282b0f53cb
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FreeGrid
  * OpenDark 2023.10.30
  * A2plus 0.4.60n
  * BIM 2021.12.0
  * woodworking 0.21.30486
  * Rocket 3.2.3
  * FreeCAD_SheetMetal 0.3.1
  * freecad_metal_workbench 0.0.1
  * kicadStepUpMod 10.20.7
  * Nodes 0.1.36
  * Defeaturing 1.2.1
  * offline-documentation 1.0.0-alpha
  * PieMenu
  * Help 1.0.3
  * FreeCAD_FastenersWB 0.4.68
  * lattice2 1.0.0
  * Assembly4 0.50.4
  * Assembly3 0.12.0
  * freecad.gears 1.0.0
  * Render 2023.8.18
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Same problem occurs in latest FreeCAD 0.22.dev

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-06:
@yorikvanhavre do we want to show deleted files, for how long and how many ?

hasecilu, 2023-11-07:
I found this in the source code `src/gui/CommandsStd.cpp` L145, it seems that deleted files should not be shown, bug?:
```c++
/**
 * Opens the recent file at position \a iMsg in the menu.
 * If the file does not exist or cannot be loaded this item is removed
 * from the list.
 */
void StdCmdRecentFiles::activated(int iMsg)
{
    auto act = qobject_cast<RecentFilesAction*>(_pcAction);
    if (act) act->activateFile( iMsg );
}
```

luzpaz, 2023-11-07:
Nice catch @hasecilu

tmksw, 2023-11-10:
@hasecilu Method `StdCmdRecentFiles::activated` is responsible for opening the recent file (MRU) which has already been added to the MRU list. I believe the fix need to be implemented when the list is propagated, it should be checked if file exists before adding it to MRU list in a marked place:

[image]

tmksw, 2023-11-11:
So I raised a PR and am not sure how to proceed from here. How do I get reviewers to look into that (minimal) changes ?

luzpaz, 2023-11-13:
@tmksw Monday is when many merges occur.

luzpaz, 2023-11-13:
Thank you @tmksw for the fix and @hasecilu for the report!

tmksw, 2023-11-13:
Thank you, I hope more PRs will follow ;)

PR #11342 (2023-11-13): Gui: Do not add non-existing files to open recent menu
It might happen that files stored in MRU list do not exist anymore e.g. have been deleted by the user from the file system.
In such a case they shouldn't be added to "Open recent" menu.

This PR fixes #11247