FreeCAD/FreeCAD #11258: [Problem] Sketcher toggle icons are not homogeneous.

PaddleStroke, 2023-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sketcher has 3 toggle icons :
[image]

The style of these icons should be uniform to convey the fact that they are toggles. Yet we have 3 different icons styles which make little sense. On top of that construction geo is easily confused with carbon copy.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-03:
Here's a proposal : 
[image]

pierreporte, 2023-11-03:
I think that your icons are less clear than what is currently in use. The new ones feel like operations on points, which would seem strange to someone ignoring that FreeCAD has only one kind of point (there is only construction points), which by the way should be improved (see my comments in #6884 #10100).

Why not use text? The problem with icons is that you can’t always find a good representation of the performed action. If FreeCAD is going to have a ribbon interface at some point (I’ve read it somewhere, not just a proposal nor a plugin), text buttons will be used (and it would solve other UI issues like worbench selector or disabled workbenches to name a few). For now, we could have “Toggle X / Y / Z” in place of icons. To display the status, maybe it could be made like `Std_TreeSyncView` that has a visual indication (provided by Qt) that the command is active.

Speaking as a professional CAD user, icon-only interfaces like FreeCAD and Catia are far less discoverable than programs with ribbon interfaces (basically every other). It is harder to guess what buttons do. With text buttons it is often just written.

FlachyJoe, 2023-11-03:
@PaddleStroke I agree your proposal is not clear. What about adding a same sign on the three icons? Or giving them two states?

FlachyJoe, 2023-11-03:
@pierreporte the ribbon vs toolbar should always be a user choice. IMHO the tooltips are sufficient to explain the usage.

PaddleStroke, 2023-11-03:
@FlachyJoe check on the forum post, there has been many proposal.

kadet1090, 2023-11-03:
Let's not make this topic about ribbon UI, we won't be able to do that any time soon. If we want to discuss such change it should be done in another issue. OnShape does not use ribbon and it is quite important player in CAD area.

As for the icons I think that flat icons that @obelisk79 is maintaining currently do a great job in that regard. 

[image]