FreeCAD/FreeCAD #11259: [Problem] Concatenated strings cannot be set as aliases in spreadsheets

rljoyner, 2023-11-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using EasyAlias or AliasManager a cell containing a calculated value will not work.

### Full version info

```shell
Example: Cell B8 contains the formula =str(C$2) + <<_>> + str($A8)
Cell C2 contains <<M6x3>>
Cell A8 contains <<Hole_Diameter>>
The cell B8 displays <<M6x3_Hole_Diameter>>

Run EasyAlias on cell B8
I would expect the alias for cell C8 to be set to <<M6x3_Hole_Diameter>>
The following error is displayed in an Error Pop-up:
'Unable to set alias $A8 at cell C8
in spreadsheet Magnets#Spreadsheet.
Remember, aliases cannot begin with a numeral or an underscore or contain any invalid characters.'
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

I think this used to work correctly in version 19 (linux) but I have not been able to verify.
I am currently using FreeCAD_0.21.1-2023-08-31-conda-Linux-x86_64-py310.AppImage.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-03:
This issue should be addressed to the authors of the mentioned macros.

The spreadsheet API has several methods to get the value from a cell.
`obj.getContents("B8")` => `'=str(C$2) + <<_>> + str($A8)'`
`obj.get("B8")` => `'M6x3_Hole_Diameter'`
The only caveat is that `get()` fails on an empty cell.

Suggest you start a forum topic and ping the authors. I am closing this.

closed 2023-11-03 by Roy-043

luzpaz, 2023-11-06:
CC @mwganson

rljoyner, 2023-11-06:
> I have informed TheMarkster via forum PM.

I fixed the issue by replacing getContent() with get() in the existing macro.
Thanks for your help.

luzpaz, 2023-11-06:
@rljoyner which macro are you referring to ?

rljoyner, 2023-11-06:
EasyAlias 
See line 161
Replace
contents = spreadsheet.getContents(selectedCell)
with
contents = spreadsheet.get(selectedCell)
 
I would imagine there is a similar issue with macro Alias_Manager, but since I got EasyAlias working I didn't pursue it.

mwganson, 2023-11-06:
I have updated the macro to incorporate Roy's idea to use get() instead of getContents().  First, getContents() is used to ensure the cell is not empty, and then get() is used to fetch the contents.  This avoids need for try: except: block since get() will throw an exception if the cell is empty.  New version: 0.2023.11.06

mwganson, 2023-11-06:
I would also note that this did formerly work in a previous version, but that version had other issues related to spreadsheet selection.  The fix for that other issue introduced this new limitation, which now has hopefully been resolved.

mwganson, 2023-11-06:
EasyAlias is what I updated.  You might need to have addon manager refresh the local cache to get the updated version.

rljoyner, 2023-11-06:
EasyAlias appears well after updating macros.

luzpaz, 2023-11-07:
Thanks! Can we close this ticket ?

rljoyner, 2023-11-07:
Fine with me. Do I need to do anything?

luzpaz, 2023-11-07:
No, not necessary. Thanks for filing the ticket.