FreeCAD/FreeCAD #11269: [Feature Request] New option: Zoom In -> relative to cursor, Zoom Out -> ignore cursor

P-St, 2023-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
thanks for this awesome product!

When zooming in or out, FreeCAD already has a feature to zoom at the mouse cursor:

[image]

However, the most intuitive way of zooming is not supported:
Zoom In -> Zoom relative to cursor
Zoom Out -> Ignore cursor

Maybe it is just me who is missing this feature - I think adding it would make working with the sketcher much more comfortable

### Full version info

```shell
[code]
OS: Windows 11
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * lattice2 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-11-14:
what other software has this type of zoom behavior?

P-St, 2023-11-15:
@adrianinsaval  Sorry, I don't know. It just came to my mind when I worked on a sketch where I had to frequently zoom into two areas, which are far away from each other. I think with this new feature you can zoom into different areas much quicker and easier, without the need to use the keyboard. However, this is just in my imagination, so I might be wrong.

adrianinsaval, 2024-03-05:
I don't see a genuine problem described here... adding this would be preferences bloat IMO.

closed 2024-03-05 by adrianinsaval