FreeCAD/FreeCAD #11270: Treeview element not readable with dark theme

arkadesOrg, 2023-11-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The treeview element for parts looks like this and is not readable.

[image]

### Full version info

```shell
[code]
OS: Archlinux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2023-11-04:
Already reported here: https://github.com/FreeCAD/FreeCAD/issues/7623#issuecomment-1287544999

arkadesOrg, 2023-11-04:
oh how did you find that? I tried... :( sorry

luzpaz, 2023-11-05:
Keep this ticket opened. I've moved #7623 to a discussion format because it was difficult to read and became unmanageable. The direct link to the discussion is https://github.com/FreeCAD/FreeCAD/discussions/11276#discussioncomment-7478894

MisterMakerNL, 2023-11-05:
You tried turning it on and off again? ;)
This color is set by preference pack and needs a reboot to be applied.

luzpaz, 2023-11-05:
We don't prompt the user to reboot FC, right ?

arkadesOrg, 2023-11-05:
> We don't prompt the user to reboot FC, right ?

I was prompted to reboot after installing a preference pack, yes.

luzpaz, 2023-11-05:
And you rebooted ?

luzpaz, 2023-11-07:
...

luzpaz, 2023-11-08:
@arkadesOrg ping. Clarrifying if you infact rebooted FC after making the change ?

luzpaz, 2023-12-05:
Closing as no response from OP

closed 2023-12-05 by luzpaz