FreeCAD/FreeCAD #11274: Freecad 0.21.1 no abre windows 10

Tecorauto, 2023-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

He instalado Freecad 0.21.1 estable en windows 10 X64 y no abre

### Full version info

```shell
Freecad 0.21.1 no arranca en Windows 10 X64, solamente aparece la presentación del arranque pero no abre el programa
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-05:
Bug rewritten in english. Please populate the ticket with appropriate information. 

1. Full [about](https://wiki.freecad.org/About) info
2. Any output that you see. please add to ticket
3. Step-by-step way for us to reproduce
4. The location name of the place you installed FC to

Roy-043, 2023-11-05:
Please check this forum post:
https://forum.freecad.org/viewtopic.php?t=30573

luzpaz, 2023-11-06:
@Tecorauto please respond

luzpaz, 2023-11-08:
No response from OP. Closing

closed 2023-11-08 by luzpaz