FreeCAD/FreeCAD #11277: DAE / Collada import ignores colors of library_materials and library_effects

strongly-typed, 2023-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am opening colored dae files from [here](https://github.com/ros-industrial/kuka_experimental/blob/melodic-devel/kuka_kr210_support/meshes/kr210l150/visual/link_3.dae) in FreeCAD

The macOS file preview is fine:

[image]

FreeCAD ignores the colors defined in the dae file:
[image]

Honestly, Blender does not support more than one color:
[image]

I checked the import setting but did not find any info about colors:
[image]

The file was generated 9 years ago with SketchUp.

### Full version info

```shell
[code]
OS: macOS 14.1
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2023-11-08:
Looks like there is more than one way to specify color in a .DAE file; maybe has something to do with using material libraries rather than direct reference.  In any case, updating the parsing code to also recognize the colors in the example file ought to do it.  I think it would go near here:

https://github.com/FreeCAD/FreeCAD/blob/0d221c0969989c5610f213d4ecc7d4bf37c1be29/src/Mod/Arch/importDAE.py#L143-L147

luzpaz, 2023-11-08:
Maybe worth CCing @davesrocketshop JIC since there is a re-write of the Materials code in progress right now.

yorikvanhavre, 2023-11-10:
Indeed those "material effects" are not supported I think. Would be interesting to see if they are accessible by PyCollada? If yes then it might be simple to implement on our side...

davesrocketshop, 2025-01-28:
It should be possible to handle this, but it's not a high priority. This would be a good first project for someone who wants to dig into the file format

luzpaz, 2025-01-28:
Per copilot:

To extract a DAE format color from `library_materials` and `library_effects` using PyCollada, you can follow these steps:

```python
from collada import Collada

# Load the DAE file
dae = Collada('path/to/your/file.dae')

# Access library_materials
materials = dae.materials

# Access library_effects
effects = dae.effects

# Function to extract color from a material
def extract_color(material):
    effect = material.effect
    if hasattr(effect, 'diffuse'):
        return effect.diffuse
    return None

# Iterate through materials and extract colors
for material in materials:
    color = extract_color(material)
    if color:
        print(f"Material: {material.id}, Color: {color}")

# Alternatively, you can directly access the effects if needed
for effect in effects:
    if hasattr(effect, 'diffuse'):
        color = effect.diffuse
        print(f"Effect: {effect.id}, Color: {color}")
```

This script will print out the color information for each material and effect in the DAE file. Make sure to replace `'path/to/your/file.dae'` with the actual path to your DAE file.

closed 2025-06-30 by chennes

bramburn, 2025-06-30:
> completed via #22083 <https://github.com/FreeCAD/FreeCAD/pull/22083>.

Damn how are you closing out so man by tickets so quick

Regards,

B.Ramburn BSc (Hons) LLM MCIArb
Snr. Quantity Surveyor & Director at Icelabz