FreeCAD/FreeCAD #11284: [Problem] Sketcher Tool Overlay: Cycle through measurements with Tab-key

maxwxyz, 2023-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@PaddleStroke how is the new overlay of the tools in sketcher called?
When sketching I can change to the other dialog to edit the other dimension but I cannot cycle back to define the other dimension again. It should me allow to switch between all present overlay dialogs with the tab key.
For example I like to edit the angle first and then the length, thats currently not possible:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git)
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-06:
Yes this is a bug, I got it fixed already on my dev branch. I will push PR soon.

maxwxyz, 2023-11-06:
@PaddleStroke in the triangle tool it already works

closed 2023-11-07 by abdullahtahiriyo