FreeCAD/FreeCAD #11290: [Feature Request] Sketcher: Add on view parameters for other tools

maxwxyz, 2023-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Ad on view parameters for other sketcher tools:

- [ ] ellipses / arcs
- [ ] circles / arcs
- [ ] polyline
- [ ] slot / rounded slot
- [ ] extend edge
- [ ] fillet / chamfer
- [ ] b-splines

@PaddleStroke I hope it is okay to create a FR for multiple tools as otherwise I would be spamming new issues.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git)
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-12-24:
@PaddleStroke if possible, the Polyline could also use the tool widget in the task dialog to display and change the current mode (line, arc, tangent, ...) of the Polyline (M). Additionally, the OVP should be displayed (line: angle, length (+maybe toggle for height/width); arc: radius + angle,...)
[image]

I've looked in the code but the LineSet seems very different from line or rectangle :D

PaddleStroke, 2024-01-02:
If you know python or other language, you should be able to navigate enough I think :) Just following patterns of other DSHs.
If you need some help don't hesitate to reach out

maxwxyz, 2024-01-14:
@PaddleStroke 
> If you're willing to tackle the missing tools OVP implementation, I would suggest starting with arc of ellipse. Because you can use other DSH as reference.

I struggle with this arc of ellipse with OVP or am I missing something:
- currently the ellipse tool has OVP but does not create a constraint for them (you enter all but no dimension constraints are created)
- it has more inputs than arc of circle or ellipse
- considering the length of the ArcOfEllipse: should this also be in degree (like arc of circle)? If so, there currently is no constraint for an angle using an ellipse

maxwxyz, 2024-01-14:
also noticed that
- 3-point-circle creates no constraints when entering all OVP
- 3-point-arc is not fully constrained (no radius constraint / third point)
- polygon tool creates no angle constraint from OVP angle
- arc slot is not fully constrained when entering all OVP. No constraint for the starting angle

PaddleStroke, 2024-01-14:
I don't remember precisely for ellipse, but the idea is that if no
constraints can be added, then none are. For instance for 3 point circles,
many OVP can't be converted to constraint.
For ellipse I don't remember precisely I need to check.

Arc of ellipse need more steps and more OVP than arc or ellipse indeed.

maxwxyz, 2024-02-16:
I've also created individual issues to keep track of the remaining ones. That's why I'm closing this one here.

closed 2024-02-16 by maxwxyz

MiqCG, 2024-06-12:
> I've also created individual issues to keep track of the remaining ones. That's why I'm closing this one here. And I've stolen your screenshot to demo OVP/widget, I hope that's ok @PaddleStroke

Dear, @maxwxyz ,

I have searched the individual issues derived of this one and I was able to find the Polyline one #12462 and Arc of Ellipse, Hyperbola, Parabola #12463.
I think is specially interesting the one about Chamfers and Fillets. Is there an issue created?