FreeCAD/FreeCAD #11291: [Problem] Sketcher: Offset on geometry with arcs and sharp angles not working

maxwxyz, 2023-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@PaddleStroke Offsetting profiles with arcs and following (non tangent) lines with sharp angles produces an error while trying to create an outer offset with type **intersection** only.

Here is a demo file with two sketches which cannot be offset with a positive value and mode intersection
[outer-offset-fail.zip](https://github.com/FreeCAD/FreeCAD/files/13265382/outer-offset-fail.zip)

[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git)
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-06:
Intersection mode is not working well in all situations. But this is a OCC limitation.

kadet1090, 2023-11-06:
It's not possible that it works in Part WB with the same geometry, logic used in both cases is exactly the same. OCC does not handle such cases and that's it. Vector graphics software handles sharp angles by beveling them and not doing intersection. 

In my implementation for #10538 I had the same issues. Hopefully I'll be able to work around them and come up with some `BRep_2dOffestBuilder` class that can handle such cases. I intend to reuse the same logic for both thin walls feature and offset tool in sketcher.

PaddleStroke, 2023-11-06:
I wouldn't say it's not possible as the implementation is not exactly the same. There's maybe something different. For example in sketcher offset I used only occ 'Pipe' and not skin or recto verso as those seems to be more buggy. But perhaps they can be interesting.

maxwxyz, 2023-11-22:
@PaddleStroke on macOS it currently crashes when trying to offset a circle. When placing the offset it closes with a segmentation fault. Regardless of offset constraint is activated or not.
could be related to https://github.com/FreeCAD/FreeCAD/issues/11313

PaddleStroke, 2023-11-22:
I don't have a mac to test. Can you confirm, it crashes only when offseting a circle? Other geometries work  correctly?

xtemp09, 2023-11-23:
@maxwxyz, could you provide the crash log and the backtrace?

maxwxyz, 2023-11-23:
ok it seems all geometries offsets crashes FreeCAD on my side.
The copy to clipboard feature of the about info seems also not to work. I will check with the next weekly build!

maxwxyz, 2023-11-24:
cannot reproduce on current weekly build. I close this issue, as pipe offset is the limiting factor. In PartWB the same profile can be offset but using the other offset modes. 

```
OS: macOS 13.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35176 (Git)
Build type: Release
Branch: main
Hash: dcd533af4bf0cb3dbc50c7d0e2fb579cad168819
Python 3.10.13, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * OpenDark 2023.10.30
  * CfdOF 1.24.8
  * Defeaturing 1.2.1
  * Manipulator 1.5.0
  * fasteners 0.4.70
  * CurvedShapes 1.0.5
  * sheetmetal 0.3.12
  * 3D_Printing_Tools
  * Curves 0.6.14
```

closed 2023-11-24 by maxwxyz