FreeCAD/FreeCAD #11305: Regression(?) in commit  #acbf135b784, internal sketcher angel calculation fails on units

berberic2, 2023-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This commit causes problems, because it does not honor units correctly leading to error-messages and strange behaviour. Example of error-message:
```
<class 'ArithmeticError'>: Quantity::operator -(): Unit mismatch in minus operation
in expression: 180 - atan(0.03)
in property binding 'Luer#Sketch.Constraints'
<Sketch> SketchObject.cpp(9395): Failed to recompute Luer#Sketch.ExpressionEngine: Quantity::operator -(): Unit mismatch
in minus operation
in expression: 180 - atan(0.03)
in property binding 'Luer#Sketch.Constraints
```
The commit https://github.com/FreeCAD/FreeCAD/pull/10743 determines if an expression has a proper unit by searching for ‘°’, ‘deg’ or ’rad’. This does not work in some cases. Example: `atan(0.03)` returns an angle with proper unit, even there is no ‘`°`’, ‘`rad`’ or ‘`deg`’.
also see: https://forum.freecad.org/viewtopic.php?t=82458 (german)

@howie-j 

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34878 +2 (Git)
Build type: Release
Branch: Meins
Hash: f0b36b0fcac6bf9b7318c14b006146effe97819b
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.14
  * Assembly4 0.50.4
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.3.1
  * Silk 0.1.5
  * CurvedShapes 1.0.5
  * DynamicData 2.53.0
  * FreeCAD_assembly3 0.11.3
  * fcgear 1.0.0
  * ExplodedAssembly
  * A2plus 0.4.60n
  * fasteners 0.4.70
  * Render 2023.8.18
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-11-07:
Thanks for reporting the issue.

I did not account for atan(), and i will try to create a fix as soon as i have time.

Do you know any other units that should be included?

berberic2, 2023-11-07:
Obviously the other inverse trigonometric functions: asin(), acos() and atan2() return angles.
And of course every value from a spreadsheet, from the DynamicData WB or a sketcher-constraint can be an angle. Maybe other, I overlooked, too…

So I’m not sure this is the right way, to do this.

howie-j, 2023-11-07:
Ok, so this is what i've found out.
Fixing the issue with asin(), acos() etc is trivial.

**The main issue** is when the expression contains external references, or has external references. See the following screencast:
Sketch 2 is a carbon copy of sketch 1, with 10 degree offset for visibility. When changing the position of the angle constraint, the whole expression changes (as you would assume). However, all other constraints with reference to the initial (now reversed) constraint, will have the wrong value now.

[Screencast from 2023-11-07 19-47-54.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/a2867200-353e-4d6c-9244-e39cfb222972)

### Possible solutions
I can think of two solutions (keep in mind i have almost zero sketcher code knowledge):

1. Only allow `reverseAngleConstraintToSupplementary()` when **creating** new angle constraints with `sketcher_dimension`, never after they are created. I guess this is the most robust solution.
2. Add a `bool reversed` state variable to the angle constraint instead of modifying the expression directly. Unsure if this is hard or even possible.

In my opinion, the most important usecase for this feature is during constraint creation, not after the constraint is made. If we can save code complexity and possible bugs by doing so, i think that would be smart.

@PaddleStroke @abdullahtahiriyo please check out this issue, and advice on a solution.

PaddleStroke, 2023-11-08:
Or alternatively we only disable reverseAngleConstraintToSupplementary for constraints that have an expression. This should be easily possible and would cover more cases than disabling for all constraints.

Ideally if we could disable only for constraint that have expression that are referenced somewhere else? But this I don't know if its possible.

howie-j, 2023-11-08:
> Or alternatively we only disable reverseAngleConstraintToSupplementary for constraints that have an expression. This should be easily possible and would cover more cases than disabling for all constraints.
Yes that's probably a better solution.

howie-j, 2023-11-29:
> Or alternatively we only disable reverseAngleConstraintToSupplementary for constraints that have an expression. This should be easily possible and would cover more cases than disabling for all constraints.

I've tested this a bit more, and i don't believe it will work.
If you have sketch 1 without expressions, but sketch 2 is a carbon copy of sketch 1, sketch 2 will get messed up when you rotate an angle constraint in sketch 1.

> Ideally if we could disable only for constraint that have expression that are referenced somewhere else? But this I don't know if its possible.

I guess this is the only way complementary angle/reversal on existing constrains could work with the current implementation, if you are sure that only constraints that are not referenced elsewhere can be rotated.

This would again create the issue of UX standardization, where some angle constraints can be rotated to complementary but others not. It would probably not be too long before complaints and issues would start emerging, with complaints that "it only works sometimes"..

PaddleStroke, 2023-11-30:
So I checked. You are correct, in the original sketch the constraint has nothing special. It's in the carboncopied that the constraint has a formula `Sketch002.Constraints[0]`.

First, this link is not very stable as it reference to a constraint id in a sketch. What if the sketch changes of name or the constraint id changes by a deletion of another constraint?

In other term the carbon copy is not handling the situation correctly. It should create document wide custom properties to hold the values of the dimensional constraints that are being copied. As to ensure the link cannot be broken so easily.
In which case the described bug would not occur.

So I do not see why we would cripple this feature of being able to move angles around because of a design issue of carbon copy.

berberic2, 2023-11-30:
> First, this link is not very stable as it reference to a constraint id in a sketch. What if the constraint id changes after a deletion of another constraint?

You are not doing this normally, or at least you should’t. You name the constraint and reference it by this name. p.e. `Sketch002.Constraints.length`.

abdullahtahiriyo, 2023-12-09:
The correct way to link constraints is by alias. This has been discussed, but it is yet to be implemented.

I did not follow this "supplemental angle" discussion/PR/merge. If you cannot fix it or need help, let me know.

PaddleStroke, 2023-12-09:
If carbon copy changes to use alias, then I think the problem mentioned will be gone. So I think the change to alias in carboncopy is the correct way to solve this bug.

adrianinsaval, 2024-03-02:
can't you just evaluate the expression separately and see if it comes out with a unit or not? why do you need to parse it manually to figure out the unit.

howie-j, 2024-03-05:
@adrianinsaval Thanks for bumping the issue, i had forgotten about it.
The unit expression part is not the main issue, see https://github.com/FreeCAD/FreeCAD/issues/11305#issuecomment-1799585590.

I have not looked at it further, but from what i understand there is no safe way to know if a constraint is referenced somewhere else, and thus the "hacky" way of dragging a constraint to supplementary by adding 180° - (angle) to the formula should be removed IMO. That should also fix the original issue reported.

PaddleStroke, 2024-03-06:
I disagree that the dragging to supplementary should be removed because of this. This is a very wanted feature, we are not removing it for an obscure inconvenience that would probably never happen in real life.

The way of fixing it is that instead of referencing to the constraint value directly, the carbon copy creates a document property 'Length = 100mm' and set this value to the constraint in first sketch. Then the carbon copied sketch should also use the document property.
So instead of : 
- cstrA = 100mm
- cstrB = cstrA
We'd have : 
- length = 100mm
- cstrA = length
- cstrB = length.

This way if the user go change the original sketch it becomes : 
- length = 100mm
- cstrA = 180 - length
- cstrB = length.

maxwxyz, 2024-08-04:
any news on this?

howie-j, 2024-08-10:
> any news on this?

Not from me at least, this issue is unfortunately a bit out of my skill level.