FreeCAD/FreeCAD #11306: [Problem] Size of the FEM feature symbols can't be changed

FEA-eng, 2023-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

An option to change the size of the FEM feature (former "constraint") symbols would be very useful. They are often too large or too small. There is a Scale property but it doesn't change anything. And can't be lower than 1.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git)
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-11-10:
Do they currently get a random size or dependent from the feature size (line, face)? I know this as well, some are huge so they overlap other geometry and some are too small to even see in the same scene. Maybe they can get a common scene size applied firstly, to be equal and appropriately sized in reference tho the entire model.

luzpaz, 2023-11-11:
CC @marioalexis84

FEA-eng, 2024-02-26:
> Do they currently get a random size or dependent from the feature size (line, face)?

Good question. It seems that they depend on the geometrical feature size but it doesn't always work properly. With multiple force load features (notice the smallest cube):

[image]

With a single force load feature applied to multiple faces:

[image]

Apart from "smarter" automatic sizing, the "scale" option should actually do something and let the user change the size of the symbols.

I also wonder what influences their density (number of symbols). It seems to depend on the geometry size as well. Some software allows setting this.

marioalexis84, 2024-02-26:
I'm working on this. I still need to finish editing all the symbol view providers to make the scale property work properly.
The density basically depends on the size and geometry of the shape.
It would also be possible to add a density property.

FEA-eng, 2024-02-26:
> I'm working on this. I still need to finish editing all the symbol view providers to make the scale property work properly. The density basically depends on the size and geometry of the shape. It would also be possible to add a density property.

Great, it will make the visualization much better. A separate property for density would also be very useful to control the extent to which the face/edge is covered with symbols. Helps reduce the visual clutter, highlight something or focus more on the geometry.

FEA-eng, 2024-03-13:
Yeah, size-based scaling is definitely broken. I applied constraints to this 8 m long beam and saw nothing:

[image]

I thought that the symbols were not added but then I zoomed in very close:

[image]

maxwxyz, 2024-04-06:
For me https://github.com/FreeCAD/FreeCAD/issues/5768 is a duplicate to this, please confirm.
https://forum.freecad.org/viewtopic.php?f=18&t=27401

FEA-eng, 2024-04-06:
> For me #5768 is a duplicate to this, please confirm. https://forum.freecad.org/viewtopic.php?f=18&t=27401

#5768 is only about force symbols, this one is more general. But both should be resolved soon anyway.

marioalexis84, 2024-04-06:
With https://github.com/FreeCAD/FreeCAD/pull/13274 will be possible change the scale of the constraints.
The initial size of the symbols is a separate issue.

maxwxyz, 2024-04-06:
Ok I'll close the other then. Is there a separate for the initial size or will it be fixed with #13274 as well?

marioalexis84, 2024-04-09:
It should be reported separately since #13274 don't involve the initial size value.

FEA-eng, 2024-04-09:
> It should be reported separately

#7030 covers this

PR #13274 (2024-04-22): Fem: Constraint symbol scaling
This pull request fixes #11306 via a rework of the constraint symbol framework:

* A `SoMultipleCopy` node is used to create multiple copies of the symbol node and save resources (currently, there is one node for each symbol).
* The symbol geometry is read from an Open Inventor .iv file (the expected file structure is detailed in the `loadSymbol` function documentation) and is made up of the symbol used at each point on the reference shape plus an optional symbol that does not is affected by multiple copies (for example, useful for those constraints that need to  [... cut]
closed 2024-04-22 by WandererFan