FreeCAD/FreeCAD #11308: [Problem] Template offset when exporting a Techdraw drawing in PDF format

DEL-TRONIK, 2023-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When exporting a drawing in PDF format using the "Export PDF" tool in Techdraw workshop, the template undergoes an offset. It works with the 'Print all pages' tool.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git) AppImage
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.5
  * Defeaturing 1.2.0
  * ThreadProfile 1.89.0
  * fasteners 0.4.66
  * ExplodedAssembly
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-07:
Can you show a screenshot that illustrates the issue ?

DEL-TRONIK, 2023-11-07:
here is an example:

[image]

DEL-TRONIK, 2023-11-08:
Here's how to reproduce the bug:

1. create a new techdraw document
2. export the page with the File -> Export PDF (no need to draw)
3. open PDF file
4. there is a shift downwards as well as to the right (as in the example above)

What diagnosis is missing?

WandererFan, 2023-11-08:
Apparently, QPrinter sets different page margins if writing to PDF vs paper.   Fix on the way.

WandererFan, 2023-11-10:
Should be fixed by PR #11325

closed 2023-11-10 by WandererFan