FreeCAD/FreeCAD #11309: [Problem] print center marks function does not work in Techdraw workshop

DEL-TRONIK, 2023-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the Techdraw workshop, when exporting a drawing, although the 'Print center marks' function is deactivated in the Techdraw preferences (Annotation tab), they are always displayed when exporting a drawing in PDF format.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git) AppImage
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.5
  * Defeaturing 1.2.0
  * ThreadProfile 1.89.0
  * fasteners 0.4.66
  * ExplodedAssembly
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-07:
Duplicate of #11228 ?

DEL-TRONIK, 2023-11-07:
I don't think so, it's not a detailed view. The problem is that the 'Print center marks' function in the preferences does not work (even if it's unchecked it still displays the center marks, which should not be the case). I just want to display the center marks when working in Techdraw, but not when exporting the drawing.

WandererFan, 2023-11-07:
[](https://forum.freecad.org/viewtopic.php?t=82532)

WandererFan, 2023-11-08:
FYI, you can use https://wiki.freecad.org/TechDraw_CosmeticCircle instead.  It seems to put the circle in the right place.

WandererFan, 2023-11-10:
Should be fixed by PR #11325

closed 2023-11-10 by WandererFan