FreeCAD/FreeCAD #11310: [Problem] In techdraw problem with Add cosmetic circle tool

DEL-TRONIK, 2023-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the techdraw workshop, when using the 'Add cosmetic arc' tool, it is not positioned correctly, here's a gif example (note that the Left view is a section view):

[image]

I also attach the previous file:
[LED_Osram_GW_CSBRM3.HW.zip](https://github.com/FreeCAD/FreeCAD/files/13285006/LED_Osram_GW_CSBRM3.HW.zip)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git) AppImage
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.5
  * Defeaturing 1.2.0
  * ThreadProfile 1.89.0
  * fasteners 0.4.66
  * ExplodedAssembly
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-11-08:
FYI, you can use https://wiki.freecad.org/TechDraw_CosmeticCircle instead. It seems to put the circle in the right place.

WandererFan, 2023-11-08:
Left is rotated -90.  The points returned by getProjVertexByIndex() are not rotated.  So the points need to be rotated to match the view.

DEL-TRONIK, 2023-11-09:
yes indeed, it works (it's my current workaround) but it forces me to manually specify the corresponding radius, which isn't necessarily ideal. Normally this works perfectly with the 'Add cosmetic arc' tool, I've already done this several times in previous versions of freecad with the same design.

WandererFan, 2023-11-11:
should be fixed by PR #11310

DEL-TRONIK, 2023-11-14:
Thank you very much @WandererFan , it works again with the 35024 (git) development version.

closed 2023-11-14 by DEL-TRONIK

luzpaz, 2023-11-14:
Thanks @WandererFan for the fix and @DEL-TRONIK for the report!