FreeCAD/FreeCAD #11311: [Problem] Assembly - Joint is not JSON serializable

howie-j, 2023-11-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a document containing an assembly and joints are saved, the following error is reported:
```
PropertyPythonObject::toString(): failed for <class 'JointObject.Joint'>
Traceback (most recent call last):
  File "/usr/lib64/python3.11/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
```

When reopening the file, the joint viewprovider is not loaded / joint is not visible and greyed out in the treeview. The following error is reported:

```
Loading Assembly workbench...
Initializing Assembly workbench...
Assembly workbench loaded
Traceback (most recent call last):
  File "/home/howie/git/fc_build/Mod/Assembly/JointObject.py", line 276, in attach
    self.draw_style.lineWidth = self.axis_thickness
                                ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'ViewProviderJoint' object has no attribute 'axis_thickness'
Traceback (most recent call last):
  File "/home/howie/git/fc_build/Mod/Assembly/JointObject.py", line 383, in updateData
    self.switch_JCS1.whichChild = coin.SO_SWITCH_ALL
    ^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'ViewProviderJoint' object has no attribute 'switch_JCS1'
Traceback (most recent call last):
  File "/home/howie/git/fc_build/Mod/Assembly/JointObject.py", line 391, in updateData
    self.switch_JCS2.whichChild = coin.SO_SWITCH_ALL
    ^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'ViewProviderJoint' object has no attribute 'switch_JCS2'
Traceback (most recent call last):
  File "/home/howie/git/fc_build/Mod/Assembly/JointObject.py", line 427, in getIcon
    if self.app_obj.getPropertyByName("JointType") == "Fixed":
       ^^^^^^^^^^^^
<class 'AttributeError'>: 'ViewProviderJoint' object has no attribute 'app_obj'
```

### Full version info

```shell
OS: Fedora Linux 38 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34935 (Git)
Build type: Release
Branch: main
Hash: e235a1b795567a99d67e1887a1c7151e60ebe395
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * QuickMeasure 2022.10.28
  * OpenDark 2023.10.30
```
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

@PaddleStroke 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-15:
This is fixed in #10764

howie-j, 2023-11-15:
Great, thanks. This will be closed when that merges then :+1:

PR #10764 (2024-02-13): [Assembly] Solve the assembly (Card 9).
Adds AssemblyObject and viewer in cpp that handle the solving by connecting to ondsel solver.
Fixes #10753
Fixes #10749
Fixes #10754
Fixes #10479
Fixes #11311
Fixes #10751
Fixes #11397
Fixes #10815
Fixes #11606
Fixes #10752
closed 2024-02-13 by chennes