FreeCAD/FreeCAD #11313: 2d offset tool GpuChannelMsg_CreateCommandBuffer MacOS

ben-h-a, 2023-11-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using sketcher 2d offset tool, on setting the value of the offset I get a segmentation fault: 

[31711:18179:1107/235144.553922:ERROR:command_buffer_proxy_impl.cc(141)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
FALLBACK (log once): Fallback to SW vertex for line stipple
FALLBACK (log once): Fallback to SW vertex processing, m_disable_code: 2000
FALLBACK (log once): Fallback to SW vertex processing in drawCore, m_disa

[image]
ble_code: 2000

### Full version info

```shell
OS: macOS 14.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34935 (Git)
Build type: Release
Branch: main
Hash: e235a1b795567a99d67e1887a1c7151e60ebe395
Python 3.11.0, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.20.9
  * Assembly4 0.50.4
```
```

### Subproject(s) affected?

None

### Anything else?

Attached is the project file zip

Steps to reproduce: 

1. launch tool and open project
2. open sketch
[2d_offset_exception.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13290632/2d_offset_exception.FCStd.zip)

3. open PROBLEM_COMPONENT > PROBLEM_BODY > PROBLEM_SKETCH
4. Highlight all geometry
5. (z, o) to launch offset tool
6. set any value for offset

Exception at file: src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h:178

Exception call stack

`SketcherGui.so!SketcherGui::StateMachine<SketcherGui::StateMachines::OneSeekEnd>::isState(SketcherGui::StateMachines::OneSeekEnd) const (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Mod/Sketcher/Gui/DrawSketchDefaultHandler.h:178)
SketcherGui.so!SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset, SketcherGui::StateMachines::OneSeekEnd, 0, SketcherGui::OnViewParameters<1, 1>, SketcherGui::ConstructionMethods::OffsetConstructionMethod>::afterHandlerModeChanged() (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Mod/Sketcher/Gui/DrawSketchController.h:378)
SketcherGui.so!SketcherGui::DrawSketchControllableHandler<SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerOffset, SketcherGui::StateMachines::OneSeekEnd, 0, SketcherGui::OnViewParameters<1, 1>, SketcherGui::WidgetParameters<0, 0>, SketcherGui::WidgetCheckboxes<2, 2>, SketcherGui::WidgetComboboxes<1, 1>, SketcherGui::ConstructionMethods::OffsetConstructionMethod, true>>::onModeChanged() (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Mod/Sketcher/Gui/DrawSketchControllableHandler.h:165)
SketcherGui.so!non-virtual thunk to SketcherGui::DrawSketchControllableHandler<SketcherGui::DrawSketchDefaultWidgetController<SketcherGui::DrawSketchHandlerOffset, SketcherGui::StateMachines::OneSeekEnd, 0, SketcherGui::OnViewParameters<1, 1>, SketcherGui::WidgetParameters<0, 0>, SketcherGui::WidgetCheckboxes<2, 2>, SketcherGui::WidgetComboboxes<1, 1>, SketcherGui::ConstructionMethods::OffsetConstructionMethod, true>>::onModeChanged() (Unknown Source:0)
SketcherGui.so!SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset, SketcherGui::StateMachines::OneSeekEnd, 0, SketcherGui::OnViewParameters<1, 1>, SketcherGui::ConstructionMethods::OffsetConstructionMethod>::finishControlsChanged() (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Mod/Sketcher/Gui/DrawSketchController.h:444)
SketcherGui.so!SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset, SketcherGui::StateMachines::OneSeekEnd, 0, SketcherGui::OnViewParameters<1, 1>, SketcherGui::ConstructionMethods::OffsetConstructionMethod>::onViewValueChanged(int, double) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Mod/Sketcher/Gui/DrawSketchController.h:290)
libQt5Core.5.15.8.dylib!___lldb_unnamed_symbol16958 (Unknown Source:0)
libFreeCADGui.dylib!Gui::EditableDatumLabel::valueChanged(double) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/build/src/Gui/FreeCADGui_autogen/include/moc_EditableDatumLabel.cpp:133)
libFreeCADGui.dylib!Gui::EditableDatumLabel::startEdit(double, QObject*, bool)::$_1::operator()(double) const (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Gui/EditableDatumLabel.cpp:171)
libFreeCADGui.dylib!QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<double>, void, Gui::EditableDatumLabel::startEdit(double, QObject*, bool)::$_1>::call(Gui::EditableDatumLabel::startEdit(double, QObject*, bool)::$_1&, void**) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/.conda/freecad/include/qt/QtCore/qobjectdefs_impl.h:146)
libFreeCADGui.dylib!void QtPrivate::Functor<Gui::EditableDatumLabel::startEdit(double, QObject*, bool)::$_1, 1>::call<QtPrivate::List<double>, void>(Gui::EditableDatumLabel::startEdit(double, QObject*, bool)::$_1&, void*, void**) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/.conda/freecad/include/qt/QtCore/qobjectdefs_impl.h:256)
libFreeCADGui.dylib!QtPrivate::QFunctorSlotObject<Gui::EditableDatumLabel::startEdit(double, QObject*, bool)::$_1, 1, QtPrivate::List<double>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/.conda/freecad/include/qt/QtCore/qobjectdefs_impl.h:443)
libQt5Core.5.15.8.dylib!___lldb_unnamed_symbol16958 (Unknown Source:0)
libFreeCADGui.dylib!Gui::QuantitySpinBox::valueChanged(double) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/build/src/Gui/FreeCADGui_autogen/include/moc_QuantitySpinBox.cpp:306)
libFreeCADGui.dylib!Gui::QuantitySpinBox::updateFromCache(bool, bool) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Gui/QuantitySpinBox.cpp:582)
libQt5Core.5.15.8.dylib!___lldb_unnamed_symbol16958 (Unknown Source:0)
libQt5Widgets.5.15.8.dylib!QAbstractSpinBox::keyPressEvent(QKeyEvent*) (Unknown Source:0)
libFreeCADGui.dylib!Gui::QuantitySpinBox::keyPressEvent(QKeyEvent*) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Gui/QuantitySpinBox.cpp:411)
libQt5Widgets.5.15.8.dylib!QWidget::event(QEvent*) (Unknown Source:0)
libQt5Widgets.5.15.8.dylib!QAbstractSpinBox::event(QEvent*) (Unknown Source:0)
libQt5Widgets.5.15.8.dylib!QApplicationPrivate::notify_helper(QObject*, QEvent*) (Unknown Source:0)
libQt5Widgets.5.15.8.dylib!QApplication::notify(QObject*, QEvent*) (Unknown Source:0)
libFreeCADGui.dylib!Gui::GUIApplication::notify(QObject*, QEvent*) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:83)
libQt5Core.5.15.8.dylib!QCoreApplication::notifyInternal2(QObject*, QEvent*) (Unknown Source:0)
libQt5Widgets.5.15.8.dylib!___lldb_unnamed_symbol13538 (Unknown Source:0)
libQt5Widgets.5.15.8.dylib!QApplicationPrivate::notify_helper(QObject*, QEvent*) (Unknown Source:0)
libQt5Widgets.5.15.8.dylib!QApplication::notify(QObject*, QEvent*) (Unknown Source:0)
libFreeCADGui.dylib!Gui::GUIApplication::notify(QObject*, QEvent*) (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:83)
libQt5Core.5.15.8.dylib!QCoreApplication::notifyInternal2(QObject*, QEvent*) (Unknown Source:0)
libQt5Gui.5.15.8.dylib!QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) (Unknown Source:0)
libQt5Gui.5.15.8.dylib!QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (Unknown Source:0)
libQt5Gui.5.15.8.dylib!QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (Unknown Source:0)
libqcocoa.dylib!___lldb_unnamed_symbol3243 (Unknown Source:0)
libqcocoa.dylib!___lldb_unnamed_symbol3244 (Unknown Source:0)
AppKit!-[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] (Unknown Source:0)
AppKit!-[NSWindow(NSEventRouting) sendEvent:] (Unknown Source:0)
libqcocoa.dylib!___lldb_unnamed_symbol3531 (Unknown Source:0)
AppKit!-[NSApplication(NSEventRouting) sendEvent:] (Unknown Source:0)
libqcocoa.dylib!___lldb_unnamed_symbol3815 (Unknown Source:0)
AppKit!-[NSApplication _handleEvent:] (Unknown Source:0)
AppKit!-[NSApplication run] (Unknown Source:0)
libqcocoa.dylib!___lldb_unnamed_symbol3679 (Unknown Source:0)
libQt5Core.5.15.8.dylib!QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (Unknown Source:0)
libQt5Core.5.15.8.dylib!QCoreApplication::exec() (Unknown Source:0)
libFreeCADGui.dylib!Gui::Application::runApplication() (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Gui/Application.cpp:2425)
FreeCAD!main (/Users/benallen/Documents/git_repos/FreeCAD/FreeCAD/src/Main/MainGui.cpp:278)
start (Unknown Source:0)`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-11-08:
No crash here.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 +2 (Git)
Build type: Debug
Branch: main
Hash: 7ba19ed35513067c9328feaff96d3cb84a08877f
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git)
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

maxwxyz, 2023-11-22:
On mac it crashes when trying to offset a circle,  maybe that is related?

luzpaz, 2023-11-26:
@JohnOCFII  please test on macos, TIA!

JohnOCFII, 2023-11-27:
> @JohnOCFII please test on macos, TIA!

Still crashes.  

```
OS: macOS 14.1.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35081 (Git)
Build type: Release
Branch: main
Hash: 32db044caa92a2adc448850179c524b793a03c3e
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

Also, when I opened the sketch, the following information was posted in the Report View window:

```
19:13:46  pyException: Traceback (most recent call last):
  File "<string>", line 12, in <module>
  File "/Applications/Other Apps/FreeCAD-35081.app/Contents/Resources/Mod/Show/mTempoVis.py", line 480, in sketchClipPlane
    editDoc = Gui.editDocument()
<class 'NameError'>: name 'Gui' is not defined
19:13:46  ViewProviderSketch: setEdit: visibility automation failed with an error: 
```

Here is the first part of the crash dump:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [30935]
Path:                  /Applications/Other Apps/FreeCAD-35081.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [30929]
User ID:               501

Date/Time:             2023-11-26 19:32:07.3296 -0600
OS Version:            macOS 14.1.1 (23B81)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       0EC91927-E011-44A3-85BA-839C8812A8A5

Time Awake Since Boot: 360000 seconds
Time Since Wake:       17403 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   freecad [30935]

Application Specific Information:
abort() called

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x182add11c __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x182b14cc0 pthread_kill + 288
2   libsystem_c.dylib             	       0x182a24a40 abort + 180
3   libc++abi.dylib               	       0x182acc6d8 abort_message + 132
4   libc++abi.dylib               	       0x182abc7c8 demangling_terminate_handler() + 348
5   libobjc.A.dylib               	       0x1827678a4 _objc_terminate() + 160
6   libc++abi.dylib               	       0x182acba9c std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x182acea48 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 36
8   libc++abi.dylib               	       0x182ace9f4 __cxa_throw + 140
9   libFreeCADBase.dylib          	       0x104ef53c4 Base::InterpreterSingleton::runStringObject(char const*) + 192
10  SketcherGui.so                	       0x11e384c80 SketcherGui::ViewProviderSketch::onCameraChanged(SoCamera*) + 472
11  libCoin.80.0.1.dylib          	       0x109006920 SoDataSensor::trigger() + 20
12  libCoin.80.0.1.dylib          	       0x1090089dc SoSensorManager::processDelayQueue(int) + 404
13  libFreeCADGui.dylib           	       0x1060767f0 SIM::Coin3D::Quarter::SensorManager::idleTimeout() + 36
14  libQt5Core.5.15.8.dylib       	       0x1082ce83c 0x108188000 + 1337404
15  libQt5Core.5.15.8.dylib       	       0x1082b1880 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
16  CoreFoundation                	       0x182bf1c90 __CFRunLoopDoSource0 + 176
17  CoreFoundation                	       0x182bf1a00 __CFRunLoopDoSources0 + 244
18  CoreFoundation                	       0x182bf05f0 __CFRunLoopRun + 828
19  CoreFoundation                	       0x182befc5c CFRunLoopRunSpecific + 608
20  HIToolbox                     	       0x18d16c448 RunCurrentEventLoopInMode + 292
21  HIToolbox                     	       0x18d16c284 ReceiveNextEventCommon + 648
22  HIToolbox                     	       0x18d16bfdc _BlockUntilNextEventMatchingListInModeWithFilter + 76
23  AppKit                        	       0x1863cac54 _DPSNextEvent + 660
24  AppKit                        	       0x186ba0ebc -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
25  AppKit                        	       0x1863be100 -[NSApplication run] + 476
26  libqcocoa.dylib               	       0x109a7bdcc 0x109a50000 + 179660
27  freecad                       	       0x104347030 main + 4932
28  dyld                          	       0x1827990e0 start + 2360
```

luzpaz, 2023-11-27:
```
File "/Applications/Other Apps/FreeCAD-35081.app/Contents/Resources/Mod/Show/mTempoVis.py", line 480, in sketchClipPlane
    editDoc = Gui.editDocument()
<class 'NameError'>: name 'Gui' is not defined
19:13:46  ViewProviderSketch: setEdit: visibility automation failed with an error: 
```

The line in question:

https://github.com/FreeCAD/FreeCAD/blob/5a4a358f13fa6c01a8930d1792fe85f0caaaedcc/src/Mod/Show/mTempoVis.py#L474-L480

What's odd is that `Gui` is declared in
https://github.com/FreeCAD/FreeCAD/blob/5a4a358f13fa6c01a8930d1792fe85f0caaaedcc/src/Mod/Show/mTempoVis.py#L31-L32
But only if we're in graphic mode (not headless) right?