FreeCAD/FreeCAD #11326: [Problem] Infinite processing when setting helix angle to 90deg

Anthony-Gaudino, 2023-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a Part helix, setting the angle option to `90deg` or `-90deg` causes FreeCAD to run a seemingly infinite processing task that takes all PC memory.

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * ThreadProfile 1.89.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.5
  * Defeaturing 1.2.1
  * A2plus 0.4.61
  * Manipulator 1.5.0
  * fasteners 0.4.70
  * lattice2 1.0.0
  * sheetmetal 0.3.1
  * toSketch 1.0.1
  * Curves 0.6.14
  * Assembly4 0.50.4
  * Assembly3 0.12.0
```

### Subproject(s) affected?

Part

### Anything else?

Maybe users should not be allowed to create a helix with exactly 90deg or -90deg angle.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-09:
Confirmed. The error occurs when changing the Helix > Angle in the Property editor.

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34810 (Git)
Build type: Release
Branch: main
Hash: 196cbecde66a1c3a1b56b8c532ad1faedff4818b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

luzpaz, 2023-11-13:
@bgbsww interested ?

bgbsww, 2023-11-14:
@luzpaz seemed almost too easy a fix, but if I understand this correctly, this is it.

PR #11390 (2023-11-22): Prevent attempting to create an infinite helix
Fix #11326.

If I understand this correctly, you should never try to go to 90 degrees, that's an infinite request, thus the crash.  So I restricted the input values.
closed 2023-11-22 by wwmayer