FreeCAD/FreeCAD #11333: [Problem] Techdraw view does not consider body placement coordinates when positionning elements on the view.

slalancette, 2023-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you create two bodies extruding from sketches in global xy plane,  and you move one body with the move command from draft bench, this changes the placement coordinates. So in model viewer, the object changes its location.

However, when using the "insert View" command for the objects, the moved object is still in the original sketch location, as if it is not considering the placement parameters of the body.

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-11-10:
Can you provide a file where this happens?

slalancette, 2023-11-10:
Yes sure, just rename the file to bug.FCStd
[bug11333.zip](https://github.com/FreeCAD/FreeCAD/files/13320082/bug11333.zip)

slalancette, 2023-11-10:
I just see that the file was created with an older version of freecad cannot reproduce the bug.

closed 2023-11-10 by slalancette

WandererFan, 2023-11-10:
This is not a bug.

Positioning on the page is controlled by a views X,Y properties.   You have two separate views, each positioned appropriately.

If you want a view of multiple objects, with their relative position intact, you have to include all the objects in the same view. 
[image]
[bug11333_mod.zip](https://github.com/FreeCAD/FreeCAD/files/13320265/bug11333_mod.zip)

slalancette, 2023-11-10:
That it is my selection method that was bad.  I was selecting the pad inside object that is why.  Check this:
[image]
[image]

slalancette, 2023-11-10:
and the upodated file.  Correct view on page 1, buggy view on page 2: 
[bug11333_2.zip](https://github.com/FreeCAD/FreeCAD/files/13320351/bug11333_2.zip)

slalancette, 2023-11-10:
Thank you very much for your support, that was quick.