FreeCAD/FreeCAD #11335: [Problem] FreeCAD causes 100% CPU usage when colors are changed on spreadsheet.

ruvenistan, 2023-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This problem occurs with versions 0.20.2, and 0.21.0.33675 (Git) AppImage.

Since there is no way to set the spreadsheet colors via the settings menus, I set the spreadsheet colors within the spreadsheet WB itself for all cells to match the ProDark theme. With all cells selected, I right-clicked on "properties", and set custom colors for all cells.

This caused FreeCAD to freeze for about 5-10 minutes, and once the color change was made, continued to cause such freezes with any change anywhere in the project. I had no problems with FreeCAD until I changed the spreadsheet colors. I don't see a way to revert just the spreadsheet colors without resetting everything to defaults, so I opened a previous project without the spreadsheet color changes, and everything is fine.

I am running FreeCAD on a Dell T-130 PowerEdge server with an 8 core Xeon E3-1270 v5 @ 3.60GHz CPU, but only a GT-710 graphics card. I have hardware acceleration turned off because of the low-powered card.

lshw output for graphics card:
product: GK208B [GeForce GT 710]
vendor: NVIDIA Corporation

### Full version info

```shell
----------------------

[code]
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.5, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
[/code]

------------------------------------

[code]
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-10:
Is there a simple way to reproduce this ?

xtemp09, 2023-11-10:
When you select all, you select `16384×26×26 = 11 075 584` cells. If change of color of a cell requires _1_ ms, the whole change will be _184_ seconds. On my VM it requires three minutes to change their color.

ruvenistan, 2023-11-10:
To reproduce this problem, do this on a test project:

0. If none, add a spreadsheet to the project.
1. Select all spreadsheet cells by left-clicking the small square area at the top left corner of the spreadsheet border where the column and row numbers meet.
2. Right-click on any cell, and select "properties".
3. A box will appear, with the "colors" tab open.
4. Left-click the colored square to the right of the "text" listing.
5. A color grid array will appear.
6. Left-click the "..." ellipsis at the bottom-right of the color grid array.
7. A custom-color creation settings box will appear.
8. Use the various tools shown to create and change the "text" color.
9. Do steps 4-8 to change the "background" color.
10. Save the changes made to the project.
11. From now on, every action should provoke the problematic behavior within this project.

xtemp09, 2023-11-11:
I'd add if one saves the file, the output _FCStd_ will be 31 MiB. If you unpack it, the _Document.xml_ will be 1 166 MiB in size. FreeCAD requires 7.7 GiB of RAM just to open the project, it uses 10.7 GiB to change the value of a single cell.

The slowdown results from parsing all these gibibytes of the 11 million cells.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34935 (Git)
Build type: Debug
Branch: main
Hash: e235a1b795567a99d67e1887a1c7151e60ebe395
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

ruvenistan, 2023-11-11:
Maybe there ought to be a limit to the number of empty cells in the file. What good does it do to keep track of cells you aren't using? Perhaps ignore anything outside the boundaries of the array of cells used in each sheet?

The best solution is obviously to allow the user to control the colors of the spreadsheet. I haven' tried to fiddle with the ProDark module to see how to do this.

Vincenzoferrara, 2024-05-30:
to mitigate this action, after editing all cells, all those without text in it are marked as "used == false" again

luzpaz, 2024-10-18:
@xtemp09 is this still an issue on RC2 ?

xtemp09, 2024-10-20:
@ruvenistan, ☝️

luzpaz, 2025-01-25:
@ruvenistan