FreeCAD/FreeCAD #11340: [Problem] Overlay flickering when focus changes

pierreporte, 2023-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I toggle the overlay feature for all panels, their content flickers constantly as the focus changes. When the mouse hovers a different face in the 3D view or over the lines of the tree view, there is some sort of opacity change. Also the title bar of the panels becomes unreadable depending on where the focus is. Below is an animated gif that shows the problem.

My settings are the default, except that the overlay has been switched on. All workbenches do this.

[image]

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git) AppImage
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-11:
Looks similar to #11249

adrianinsaval, 2023-11-14:
Hot take but I don't think the overlay mode is well suited to being used with the awful windows default style. We should geta good stylesheet made the default and forget about this sort of problem.

luzpaz, 2023-11-14:
> Hot take but I don't think the overlay mode is well suited to being used with the awful windows default style. We should geta good stylesheet made the default and forget about this sort of problem.

https://github.com/FreeCAD/FreeCAD/discussions/11276

pierreporte, 2023-11-18:
I’ve installed the OpenDark theme. It doesn’t have the flickering so it seems to be related to the theme.

pierreporte, 2024-06-15:
The work done on themes eventually solved the issue.

```
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git) AppImage
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.18
```

closed 2024-06-15 by pierreporte