FreeCAD/FreeCAD #11343: Pressing ESC to exit contextual dimensions doesn't exit from dimensioning

JohnOCFII, 2023-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After using the contextual dimension tool, pressing the `ESC` key does not exit dimensioning, and the cursor remains showing the dimension tool.  Switching to a non-contextual dimension tool and THEN pressing `ESC` does allow me to exit dimensioning mode.

Problem exists in latest two version available from the weekly downloads page (versions 0.22.0dev.34959 and 0.22.0dev.34873) but problem does **NOT** occur in version 0.22.0dev.34810.

### Full version info

```shell
Fails on:

OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git)
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```
```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34873 (Git)
Build type: Release
Branch: main
Hash: 889c8c49f1bbc0910c8dc32a512f4c6fbedeb8b9
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

Works properly on:
```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34810 (Git)
Build type: Release
Branch: main
Hash: 196cbecde66a1c3a1b56b8c532ad1faedff4818b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

![CursorDimensioning](https://github.com/FreeCAD/FreeCAD/assets/7844212/0a052ee2-e25e-443c-9a8a-efc875337110)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-15:
I can replicate this issue.
@abdullahtahiriyo I would guess that this was introduced by your PR to better handle ESC on new DSH. Can you advise what needs to be done for legacy DSH to support the new workflow?
Thanks

PaddleStroke, 2023-11-15:
Nevermind I found the issue.

closed 2023-11-15 by abdullahtahiriyo

JohnOCFII, 2024-01-12:
@PaddleStroke - this same issue is now impacting Polyline (and perhaps others) as of the past couple of weeks' builds:

```
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35554 (Git)
Build type: Release
Branch: main
Hash: 8e2ab9f76888ac4da971f9d5069cec7bcc2b572f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
```

PaddleStroke, 2024-01-13:
Thanks for the report. I think perhaps all the legacy tools have this
issue. I will check later