FreeCAD/FreeCAD #11345: [Problem] Can't create walls in Arch: [wall name] has a null shape

jlmuir, 2023-11-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can't create walls in the Arch workbench (nor the BIM workbench).  When I try, I draw the two points, but it doesn't appear and I get a message like the following in the Report view:
```
17:26:15  Wall has a null shape
```
I look in Preferences > Arch > Defaults, and I see that the wall width is 0.00 mm.  I change it to 114.3 mm, and press OK.  I open the preferences again, and I see that it kept the wall width I set; it is not 0.  However, I try to create a wall again, and I get the same behavior where no wall shows up, and I see the Report view message that the wall has a null shape.  I open the preferences again, and the wall width is back to 0.00 mm. :-(

If it makes a difference, I have the "Building US" unit system set in Preferences > General.  I also am unable to set the Arch defaults preferences in "Building US" units; I have to enter them in mm.  But I was able to make walls in the previous version even with this limitation.

### Full version info

```shell
[code]
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * parts_library
  * A2plus 0.4.61
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jlmuir, 2023-11-11:
Interesting.  Given the title change and addition of the Units label to this issue that @Roy-043 made, I tried changing the unit system to Standard, and lo and behold, creating walls worked!  I then changed the unit system back to Building US, and once again creating walls did not work, and I got the same message about the wall having a null shape.

jlmuir, 2023-11-11:
I tried downgrading to FreeCAD 0.20.2, and creating walls works in it, and the Arch default wall width does not get reset to 0 mm.  Here's the version info for that:

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * parts_library
  * A2plus 0.4.61
  * BIM 2021.12.0
```

Roy-043, 2023-11-11:
This is indeed related to the Building US unit system. I somehow thought that you and @luzpaz had already concluded that.

If I replace the Arch and Draft folders of V0.21 with those of V0.20, the issue is still there. This problably means the issue is (at least in part) due to changes in the Core unit system.

Roy-043, 2023-11-13:
There is definitely a mistake in the Building US unit system:

0.21
`FreeCAD.Units.Quantity(200, FreeCAD.Units.Length).UserString` => `'7" + 7/8"'`
`FreeCAD.Units.Quantity('7" + 7/8"').Value` => Syntax error

0.20
`FreeCAD.Units.Quantity(200, FreeCAD.Units.Length).UserString` => `'7+7/8"'`
`FreeCAD.Units.Quantity('7+7/8"').Value` => `200.02499999999998`

adrianinsaval, 2023-11-14:
why is this dependent on the unit formatting?

Roy-043, 2023-11-15:
When the taskpanel opens a string like `'7" + 7/8"'`is placed in the Width inputbox. This triggers:
https://github.com/FreeCAD/FreeCAD/blob/e937cc1efc3e210655575a408425098fafef39e7/src/Mod/Arch/ArchWall.py#L558
And `self.setWidth` receives a zero value.

If another unit system is used `self.setWidth` is not triggered when the task panel opens.

In the user interface you cannot even enter `7" + 7/8"` in the Width inputbox.

yorikvanhavre, 2023-11-16:
Strangely, 7" 7/8" works (gets converted to 7" + 7/8" after tabbing out, which is not accepted by the input field)

adrianinsaval, 2023-11-17:
@ConnorWorrell have you got any clue on what is going on here?

ConnorWorrell, 2023-11-23:
@jlmuir can you try out this branch and let me know if it works for you: https://github.com/ConnorWorrell/FreeCAD/tree/ArchWallThicknessUnitsIssue

Also this should work with all units now:
> I have to enter them in mm.

I switched the 3 `Wall options` input boxes to quantitySpinBox, I didn't switch the Local X,Y, and Z because they get updated a lot in the "interactive mode" the extra parsing that quantitySpinBox does would likely cause a performance hit.

QuantitySpinBox dosen't currently implement the `returnPressed()` socket or anything similar, as far as I'm aware, so these lines throw many warnings.

https://github.com/FreeCAD/FreeCAD/blob/e937cc1efc3e210655575a408425098fafef39e7/src/Mod/Arch/ArchWall.py#L563-L567

Roy-043, 2023-11-23:
> why only mm units work here:

@ConnorWorrell That is not correct. AFAICT the situation is quite different: only Building US FAILS here.

Please also consider that this is an issue that affects many Draft and Arch tools. I hope you are not suggesting we 'fix' all of these tools to work around a bug in Core?

ConnorWorrell, 2023-11-24:
> AFAICT the situation is quite different: only Building US FAILS here.  I hope you are not suggesting we 'fix' all of these tools to work around a bug in Core?

This is correct. The change to Core caused this issue. But I think this issue is using the InputField widget (which has been essentially unmaintained for 10 years). It assumes the output of the ExpressionParser will be parsable by the ExpressionParser; which is not currently the case for USBuilding. There is an issue for this here: https://github.com/FreeCAD/FreeCAD/issues/9552, but there are a lot of moving parts, and all standardized ways to represent USBuilding numbers are ambiguous. For example: The United States National CAD Standard V4 callsout the format as #'-#-#/#", but should 1'-1-1/2" be evaluated like a number: 13.5" or an expression 1'-(1-1/2)"=10.5".

In the spirit of reducing duplicated input widget code (again; this isn't the first time these features have been duplicated). I propose:

1. Add the returnPressed socket to QuantitySpinbox to make it a drop in replacement.
2. Replace all the input fields with QuantitySpinBox.

Once QuantitySpinBox is a drop in replacement the swap should be trivial.
https://github.com/FreeCAD/FreeCAD/blob/e937cc1efc3e210655575a408425098fafef39e7/src/Gui/InputField.h#L54-L55

Doing this also has the benefit of pushing anyone referencing this code when making new GUIs to use documented functions, and continues my crusade to unify the quantity input widgets.

jlmuir, 2023-11-24:
> @jlmuir can you try out this branch and let me know if it works for you: https://github.com/ConnorWorrell/FreeCAD/tree/ArchWallThicknessUnitsIssue

@ConnorWorrell, thank you for working on this!  Unfortunately, I [can't get the FreeCAD dependencies to build](https://github.com/FreeCAD/homebrew-freecad/issues/434), so I can't build your branch.

luzpaz, 2023-12-01:
Bumping. We need to be able to build PRs for testing cross-platform.

goertzenator, 2025-01-16:
I just ran into this while trying to do the in-game BIM tutorial in FreeCAD 1.0.  It was a little discouraging that the very first thing you are directed to draw doesn't work, but the error message eventually pushed me to the find the width and height settings (which I had to scroll way down for).  Subsequent walls adopted the width and height of first wall and did not produce an error.

I should mention that I had an older version of FreeCAD that I briefly looked at a year ago.  I uninstalled that one after installing 1.0.

Roy-043, 2025-03-17:
@maxwxyz maybe this issue should become a blocker?

Roy-043, 2025-03-17:
FWIW:
Potential (?) workaround fix: change `+` to `--` internally:
`FreeCAD.Units.Quantity('7" -- 7/8"').Value` => 200.02499999999998

maxwxyz, 2025-03-17:
@Roy-043 we've released 1.0 with this issue already

yorikvanhavre, 2025-03-18:
I think we could do this:

* Never create a wall (with the UI) with a width, length or height of 0. Yield an error instead.
* If the user creates a wall from a selected line, or other object, or using Python, then we let them do what they want and don't stop anything. But in any case the wall's `execute()` code should be a bit more verbose when it's unable to create geometry and tell users what is happening.

I will do this as part of https://github.com/FreeCAD/FreeCAD/pull/19554 , unless someone decides to work on it first ;)

Roy-043, 2025-04-05:
> we've released 1.0 with this issue already

I understand that. Basically V0.21 and V1.0 have this problematic unit system. It would be a good idea to get it fixed in the current dev cycle. We keep on getting error reports for the Draft and BIM workbench where the Building US system is the actual culprit.

3x380V, 2025-06-28:
@Roy-043, sum of inches expressed as `7" + 7/8"` most likely never worked, however it could be rewritten as `(7 + 7/8)"`. Fixing that means to touch parser and while I know theory behind that, I wrote something simple using lexer and parser last time almost 30 years ago. I will try to ask friend of mine to help me with that, but that could take a while.

Roy-043, 2025-06-29:
> sum of inches expressed as `7" + 7/8"` most likely never worked

The problem is that in version 0.21 and version 1.0 the Building US unit system produces such strings. In 0.20 the string for the same value would be '7+7/8"'.

I have suggested an intermediate fix: replace `+` with `--` internally. If that works, and there is no better option, would you consider that?

https://github.com/FreeCAD/FreeCAD/issues/11345#issuecomment-1808135862
https://github.com/FreeCAD/FreeCAD/issues/11345#issuecomment-2730442774

3x380V, 2025-06-30:
Well, almost everything can be implemented, but looking at the history of Quantities, I'd like to have a bit better task assignment than just fix this. I do not use imperial systems at all neither I know what is common practice. Also Expression parser does not get some cases right, so it would be worth to fix it as well.

Currently FreeCAD parses both `7 + 7/8"` and `7" 7/8"` correctly, but the output has changed and it changed various times based on history. Just see how it looked when added in ad92a2df25c03805b3f782cc79a743e972f81f94 and see also 223725c; it seems temporary workaround is still here almost for a decade and `"` to the non fractional inches was added. So to avoid just another round fixing issues which are creating another issue, I'd like to know what is valid string to parse and how should returned `UserString` look like.

furgo16, 2025-07-21:
Would picking up https://github.com/FreeCAD/FreeCAD/pull/11621 help?

Roy-043, 2025-07-21:
Possibly. The easiest (workaround) fix would be to replace `+` with `--` before further processing the text input. If doing that in the unit system code is problematic, then it can be done in [InputField.cpp](https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/InputField.cpp). I do not know C++, so I can't do this myself.

furgo16, 2025-11-10:
I looked at it a while ago, but with the caveat that I did not (and I still don't) understand much the inner workings of the units system in FreeCAD. From my limited understanding, I still believe that rather than spending time on a workaround, it's worth resurrecting the unfinished attempt at https://github.com/FreeCAD/FreeCAD/pull/11621. The C++ change appears trivial enough (add the `returnPressed()` signal to make the widgets swappable), and the rest is refactoring to replace a widget with another.

However,

1. There would be some decisions to be made to redefine shortcut keys,
2. There is also another reportedly [unresolved issue](https://github.com/FreeCAD/FreeCAD/pull/11621) ("after pressing return the previous text box doesn't get deselected").
3. While I think those two should be solvable, I'm also concerned about the comment about "There is a small but noticeable delay in the wire frame preview." in the PR description => that might point to a performance issue with the new widget, which might not be trivial to address.

In any case, I fully agree about trying to tackle this one for the release. I'm always concerned to end up in an echo chamber with BIM: here is a very very capable workbench with high ponential, but only a few people can use it proficiently or develop it. Without a fix for this issue we're effectively banning a high percentage of the population to even start using it. Yes, it's risky before the release, but how worse can it get than not being able to use the workbench? Famous last words :-)

Perhaps other maintainers might like to chime in too.

closed 2025-11-14 by chennes

semhustej, 2025-11-15:
@maxwxyz shouldn't this issue stay open as the fix proposed by Roy is a workaround? If a proper solution is implemented in the future, the workaround can be removed.

pohjolam, 2026-03-19:
I'm noticing an issue very similar to this that I believe to be related, and I'm not sure if anyone has mentioned this, but BIM specifically fails by null wall shape when generating walls on a sketch that has a negative z position.