FreeCAD/FreeCAD #11349: Displacement datum plane in wrong direction

cnc50, 2023-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Created a datum plane YZ (Datumplane001). Wanted to set a displacement in x direction. Only z direction is possible. It seems that x and z are mismatched.

### Full version info

```shell
Freecad Version 0.21.1 33668
```

### Subproject(s) affected?

None

### Anything else?

[Halteplatte 1 V2.zip](https://github.com/FreeCAD/FreeCAD/files/13325392/Halteplatte.1.V2.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2023-11-11:
Displacement is relative to the object coordinates. Z is normal to the plane so vectors X and Y are coplanar, a translation in these directions result in nothing more for an infinite plane.

cnc50, 2023-11-11:
Thanks a lot.

closed 2023-11-11 by Roy-043