FreeCAD/FreeCAD #11350: [Problem] Sketcher OVP not visible

maxwxyz, 2023-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Default is hidden I can see, but when I enable them in the preferences to "dimensional only", they do not appear. I have to set them to "all" and then change back to "dimensional", to only see the dimensional ones.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git)
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-14:
I cannot replicate your issue.
And default should be 'Dimensional only'. Tried just now and seems to be correct.
Note the tool need to be stopped and restarted for the change to be taken into account. Are you sure you exited the tool in your tests?

maxwxyz, 2023-11-14:
@PaddleStroke I cannot reproduce as well. But the default is disabled.
Also when the Tab or Sketcher is reset, it goes back to Disabled:
[image]

howie-j, 2023-11-14:
I have the same issue. How to reproduce:
- Delete/rename/move the user.cfg file
- Start freecad -> standard part -> sketch -> line = no OVP
- Open preferences, press apply
- Open the sketch again
- OVP are working

While the default OVP setting is Dimension only, i guess that parameter somehow is not written before apply is pressed.

closed 2023-11-15 by abdullahtahiriyo