FreeCAD/FreeCAD #11351: [Problem] Sketcher: OVP dimensions/constraints not working anymore

maxwxyz, 2023-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@PaddleStroke something broke with OVP. Here an example for a line. I can set the OVP and they are somehow saved but the outcome is a totally different. I type in the parameters and cycle through them with tab key only:
https://github.com/FreeCAD/FreeCAD/assets/6246609/3e5ee3be-1340-44e4-bbae-ae466f2dff38

When I set the first point by clicking, then input the next OVP it places it at a different location:
https://github.com/FreeCAD/FreeCAD/assets/6246609/2af901c0-2e88-40aa-9057-c6c735399dfa

Some points seem to look for an auto constraint on the Y axis.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git)
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-12:
@abdullahtahiriyo did you made recent changes to auto-constraints system in the new architecture ?

PaddleStroke, 2023-11-12:
I just tried to replicate and cannot. @maxwxyz can you confirm you still have this on latest master?

maxwxyz, 2023-11-12:
Just build latest main on macOS and I can still reproduce. When entering OVP values and apply them with `Tab` or `Enter` it accepts them but when the final point of a line is drawn, everything snaps to the Y-axis with applied point on constraints.
[image]

When I do not enter OVP for the first point of a line, that point does not get the on Y-axis constraint applied. Only the ones where I enter a value in OVP:
[image]

```
OS: macOS 13.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34982 (Git)
Build type: Release
Branch: main
Hash: 55639df57325290e5a9d98ef48d6e17898fe0fc1
Python 3.11.0, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * CfdOF 1.24.4
  * Defeaturing 1.2.0
  * GDT
  * Manipulator 1.5.0
  * fasteners 0.4.59
  * CurvedShapes 1.0.4
  * sheetmetal 0.2.63
  * 3D_Printing_Tools
  * Curves 0.6.13
  * Assembly4 0.50.2
```

abdullahtahiriyo, 2023-11-12:
The changes I brought yesterday should not affect it, as there are mere interface improvements.

In order to reproduce, it is best that you give step by step instructions, or even better a small video, to see which autoconstraints were detected, which OVP mode you are using (all, or only dimensionals). and the exact sequence of events.

maxwxyz, 2023-11-12:
Here a quick demo:
- drawing line only with mouse, no input in OVP is working
- drawing a line and input OVP for start and end point: not working, both snap to the y-axis
- drawing a line and input only the end point in OVP: not working, end point snaps to y-axis
- drawing a line and input only the start point coordinates in OVP: not working, start point snaps to y-axis

https://github.com/FreeCAD/FreeCAD/assets/6246609/758dd916-3b6b-4389-971a-c0fb0a5062f8

And here is the final file which can be analyzed: 
[ovp-demo.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13327738/ovp-demo.FCStd.zip)

abdullahtahiriyo, 2023-11-12:
I could only do a quick test right now (and I must soon leave), line at 10, 20, length=20, angle=60:

[Peek 2023-11-12 11-53.webm](https://github.com/FreeCAD/FreeCAD/assets/5249162/43d6e7e1-993d-43df-bf98-3d6599ecd2eb)

maxwxyz, 2023-11-12:
@abdullahtahiriyo I did a fresh built on a windows machine and same behavior as on the macOS build.
Identical behavior, can reproduce as well:

https://github.com/FreeCAD/FreeCAD/assets/6246609/6bf45e84-80fc-4b7e-9cc9-cbc9794dc30b

abdullahtahiriyo, 2023-11-12:
@PaddleStroke 

I have just tried with main branch. But I cannot reproduce it.

I think it is clear from his screenshots that it is not working for him. When I (think I) do the same, I do not get that behaviour, but the expected one.

```
OS: Ubuntu 22.04.3 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34983 (Git)
Build type: Debug
Branch: main
Hash: 9bb996e2f329bb93fe96053f425ec8c8fcfc5422
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
```

maxwxyz, 2023-11-12:
@abdullahtahiriyo @PaddleStroke ha, figured something out. This happens only, when the sketch is on XZ-plane or YZ-plane.
It behaves normal on XY-plane.
Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/c030e981-ae32-4160-9a05-3eb900edd6bf

Can also be reproduced in official weekly build for Windows from 2023-11-09.

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git)
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```

maxwxyz, 2023-11-12:
@PaddleStroke it only does not work on Line and Rectangle tools. Point and Polygon do not show this behavior!

PaddleStroke, 2023-11-12:
Thank you I now can replicate this problem. This must be a problem in the addConstraints()

PaddleStroke, 2023-11-14:
@abdullahtahiriyo So I'm looking into it. The fact that it happens only when not in XY plane makes me thing this is not something I introduced. After some test it seems that the mouse position is projected on the XY plane after entering values in OVP.

Some notes along that seems related : 

```
    void finishControlsChanged()
    {
         ...
        // ensure that object at point is preselected, so that autoconstraints are generated
        handler->preselectAtPoint(lastControlEnforcedPosition);
        ...
    }
```
Why are we ensuring autoconstraints are generated when user input something? My intuition tells me we should rather ensure autoconstraints are NOT generated in this case?

PaddleStroke, 2023-11-14:
@abdullahtahiriyo So I have tracked this bug. It is hard and long to explain by writing so I made a video : https://youtu.be/0odAjH3Twyw

Long story short, `handler->preselectAtPoint(lastControlEnforcedPosition);` is the cause of the bug. It preselect H_axis for not good reason. So SnapManager snap onSketchPos to y = 0. and so...

PaddleStroke, 2023-11-14:
I let you see if what you think is best. I don't understand the point of doing `preselectAtPoint `here so I would just remove it. But maybe you have a reason in which case `preselectAtPoint `needs to be fixed.

maxwxyz, 2023-11-14:
And why does it only happen when the sketch is parallel to the global XZ-plane or YZ-plane? Do the other OVP tools like point ot polygon not call `preselectAtPoint` or why are they working and line and rectangle have this error?

PaddleStroke, 2023-11-14:
I don't know. H axis is preselected by preselectAtPoint  when not in XY plane.
The bug is also here actually for all tools. Try to start drawing a rectangle. Type 50. Enter. Then click without moving mouse. And the rectangle draws on the H axis.

As to why the autoconstraints are not screwed in the other tools I'm not sure.

maxwxyz, 2023-11-14:
True, without moving the mouse it is there. But when I do not enter a distance in OVP it works. When I enter both distances for the starting point, it applies the wrong snap constraint after I place the end point. Problem regarding I move the mouse when I input both coordinates.

abdullahtahiriyo, 2023-11-14:
preselectAtPoint would attempt a preselection at the given point. 

The reason it was introduced is because if you enter a dimensional value, but this would lead to a coincidence autoconstraint (for example). Imagine both constraints are redundant as a result. The dimensional value would have lower priority than the autoconstraint, the mechanism should then take the autoconstraint and ditch the dimensional. The preselect enables exactly this. Without it, you will not get the autoconstraint detected.

The principle is that of preferring geometry constraints over dimensional constraints.

Removing it introduces a bug. The solution should be to fix it. I will try to take a look, but I guess is simply that a placement transformation is missing.

PaddleStroke, 2023-11-14:
I see.
Yes you are right its probably a transformation missing.

abdullahtahiriyo, 2023-11-14:
https://github.com/FreeCAD/FreeCAD/pull/11400

Now the preselection works fine. However, the autoconstraint priority does not work. I do not know if this is because of code that was removed lately. This was working for lines in the old framework. But that is a separate issue...

maxwxyz, 2023-11-20:
shall I close this issue now that it is merged or should I keep it open to fix autoconstraint priority?

abdullahtahiriyo, 2023-11-22:
@maxwxyz 

If the issue is not reproducible anymore. I think this issue should be closed.

Feel free to open another issue for the autoconstraint priority. 

That is something that will need quite some taking into account lots of things. For the old architecture, I literally did sequence maps of the API to ensure this was taken into account. It is not a one day work...

maxwxyz, 2023-11-22:
@abdullahtahiriyo I'm not familiar with the autoconstraint priority or how this affects the user. I close this issue.

closed 2023-11-22 by maxwxyz