FreeCAD/FreeCAD #11352: [Problem] Sketcher: Cursor Icons are different from Toolbar Icons

maxwxyz, 2023-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The mouse cursor overlay icons (when a tool is selected) are different than the corresponding toolbar icons (for almost all tools).
In general, they are also overlapping with the displayed coordinates / dimensions next to the cursor.

A few examples for the cursor icons:
- PolyLine Tool: displays the old toolbar icon
- Arc: displays a mirrored icon of the toolbar icon
- Sketcher fillet: displays a chamfer cursor (not even a feature in sketcher), toolbar is a fillet
- 3-point circle: displays the arc toolbar icon
- Line/Splines/Circle/Rectangle: some completely different icons
- Rectangle options: Icon is squared, mouse pointer not. Rounded rectangle does not show the points

A few examples from above:
[image]
[image]
[image]
[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git)
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2023-11-17:
I think they are png files.

luzpaz, 2023-11-20:
Who wants to tackle this ?

maxwxyz, 2023-12-17:
> Who wants to tackle this ?

created a PR

PR #11749 (2023-12-22): Sketcher: adjust cursor icons to match toolbar icons
fixes #11352
adjusted the cursor / pointer icons to match the toolbar icons
- polyline was still showing the old icon
- fixed fillet icon (was a chamfer icon)
- fixed BSpline icon
- fixed trim icon
- continuous design of cursor icons in alignment with the toolbar icons
- added icons for rectangle options
closed 2023-12-22 by abdullahtahiriyo