FreeCAD/FreeCAD #11353: [Problem] Std_Transparency : Toggle transparency of objects - not working as expected.

maxwxyz, 2023-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The transparency toggle does not work as expected (https://github.com/FreeCAD/FreeCAD/pull/10805)  @PaddleStroke  
Example file: [toggle-transparency.zip](https://github.com/FreeCAD/FreeCAD/files/13325831/toggle-transparency.zip)

- If I select an item in the 3D view and toggle via right click menu it works. Selecting the body in the tree and toggle again: not working
- Selecting a group in the tree and toggle: does not work
- Selecting multiple in the tree an toggle: does work; select then some bodies in the 3D view and toggle: does not work for all selected.
- Selection order in the tree matters some times if the feature works or not

I think it might have to do with the issue: https://github.com/FreeCAD/FreeCAD/issues/11132 where the space bar hide/show is bad in PartDesign WB as it chooses the last feature when you select in the 3D view and not the body as if you select it in the tree view.

Quick demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/221aa778-eb51-4c6d-875c-994f29472349

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git)
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-11-11:
Maybe also related to the issue with the Std_ToggleObjects (which kind of does the same) not working as expected: https://github.com/FreeCAD/FreeCAD/issues/11131

PaddleStroke, 2023-11-12:
> If I select an item in the 3D view and toggle via right click menu it works. Selecting the body in the tree and toggle again: not working
> Selecting multiple in the tree an toggle: does work; select then some bodies in the 3D view and toggle: does not work for all selected.

Here the issue is that selecting in the view select the last feature of the Body. While selecting in the tree selects the body itself.

> Selecting a group in the tree and toggle: does not work

Indeed group is currently not handled.

> Selection order in the tree matters some times if the feature works or not

I guess this is the same issue as the first one.

So here I see 2 issues : 
- group is not supported. 
- For PartDesign bodies we should ignore the inner features and only toggle the body.

maxwxyz, 2023-11-12:
Yes, seems to fall back on the case when nested objects are in the selection. If this could be resolved, that always the body is selected (or toggled), this should also happen when hiding the item with spacebar (same procedure) https://github.com/FreeCAD/FreeCAD/issues/11132

PaddleStroke, 2023-12-16:
This can be assigned to me, I'll get back to it at some point. Unless someone else want to tackle it

maxwxyz, 2023-12-17:
@Roy-043 could you assign him?

PaddleStroke, 2024-01-11:
Also Roy mentioned a small problem in the placement in the menu. I'm posting here to centralized the things to solve for this tool : 
[image]

Freedman-CB1, 2024-01-11:
The expected behavior was to change the Transparency at the object level. Part and Group don't have their own transparency property so I would not expect them to be included in the selection process. I'm not sure about Arch. The idea of the toggle was a quick selection of an object/objects in the Workbench including Sketcher. Valid Sketcher selections will be from the tree only because Sketcher is sketch object modal in the 3D. Maybe the name "Transparency toggle" needs to be "Object transparency toggle" for clarity.

PR #13180 (2024-04-01): Core: Std_ToggleTransparency: Fixes #11353
Fixes #11353 : 
- Groups now toggling their content correctly.
- The command toggle the PD body instead of the tip
- The command placement in the View menu is fixed.
commit 4b16444b (2024-04-01): Core: Std_ToggleTransparency: Fixes #11353
closed 2024-04-01 by chennes