FreeCAD/FreeCAD #11354: [Feature request] Ability to Reset Toolbar and Toolbar Position

maxwxyz, 2023-11-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have not found a way to reset the toolbar contents and positions in FreeCAD, as this can become a mess when customizing them or moving them around, also on multiple monitors.
I stumbled upon a macro https://forum.freecad.org/viewtopic.php?p=718351#p718351 --> https://github.com/FreeCAD/FreeCAD-macros/pull/79 but this does only reset the position, and not even on all workbenches (e.g. not working in Sketcher WB).

I propose a button either in the customize dialog, in the View>Toolbar menu bar, or in the preferences dialog. Maybe it can be combined with the new reset option in the preferences:
[image]
- reset all toolbars with the first option
- reset the toolbars for the current selected WB with the second option

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git)
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * CfdOF 1.24.7
  * CurvedShapes 1.0.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.68
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.1
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-11-29:
@kadet1090 @PaddleStroke can this be integrated in the changes of the setting dialog / de-cluttering of toolbars?

kadet1090, 2023-11-29:
Don't think so. Toolbar positions are managed outside of system that is used for parameters and preferences (which are just fancy way to set parameters). FreeCAD has quite a lot of toolbars that are managed by different workbenches - it probably would be hard to achieve. This is however something, that'd be very beneficial to have so we probably should put at least some time to it.