FreeCAD/FreeCAD #11371: FreeCAD custom toolbars cant have subcategories [Problem]

CharlotteBeLike, 2023-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Most CAD softwares like fusion360 and onshape have customizable toolbars that allow you to have dropdown menus with more tools. FreeCAD does have this, but not in custom toolbars. I think it would be nice for users to be able to have tools with other tools nested inside them using dropdown menus.
Thank you :)

### Full version info

```shell
0.21
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-15:
Yes this is has been discussed before and it is on the roadmap.

maxwxyz, 2024-03-02:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/9926

closed 2024-03-02 by maxwxyz