FreeCAD/FreeCAD #11381: [Problem] PartDesign_Pad doesn't show the new UpToShape type in its task dialog

FlachyJoe, 2023-11-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

 #11136 introduced a new type to pad up to the base shape. The task dialog combo box doesn't show it.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35022 (Git)
Build type: Release
Branch: main
Hash: 69761e72d2002c3bc064a013600c50cf413aa311
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
```
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-11-15:
maybe it first should be discussed what types are needed in general. 
https://github.com/FreeCAD/FreeCAD/issues/11135#issuecomment-1812086962
https://forum.freecad.org/viewtopic.php?p=719123

FlachyJoe, 2023-11-15:
#11392 remove this new mode and give the ability to select shape(s) and/or face(s) in UpToFace mode.
So this issue will be closed if #11392 is merged. But another one will be open to change the task dialog behavior with UpToFace…

FlachyJoe, 2024-05-30:
@kadet1090 FYI

maxwxyz, 2024-08-25:
@FlachyJoe this can be closed, right?

closed 2024-08-25 by maxwxyz

FlachyJoe, 2024-08-26:
@maxwxyz Yes, great!