FreeCAD/FreeCAD #11391: [Problem] [Techdraw] - Add cosmetic arc tools cause problem reverse

DEL-TRONIK, 2023-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I used FreeCAD version 34387 (Git) to create a drawing with TechDraw workshop, using the 'Add cosmetic arc' tool like this:

https://github.com/FreeCAD/FreeCAD/assets/104136464/a366e5b1-e426-420c-bbdc-54a005f3ce08

Everything works perfectly, but with more recent versions of FreeCAD, when I open my drawing with the new version of FreeCAD, the drawn arc now appears on the other side (see the screen below), how come?

[image]

According to my various tests, this seems to have appeared from version 34873 (Git) of FreeCAD, but it may be earlier. Is this a bug?

Here is the file:
[LED_Osram_GW_CSBRM3.HW.zip](https://github.com/FreeCAD/FreeCAD/files/13346945/LED_Osram_GW_CSBRM3.HW.zip)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35024 (Git) AppImage
Build type: Release
Branch: main
Hash: 4ff00ef99869e35295a5a17b8f486901504c655b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.5
  * Defeaturing 1.2.0
  * ThreadProfile 1.89.0
  * fasteners 0.4.66
  * ExplodedAssembly
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-11-14:
I don't use this function much, so bear with me. 

If I make an arc from 3 points, the first point is the centerpoint, the second is the start point, and the arc is drawn counter-clockwise (conventional angle increasing) to the second point. 

If the first point is at 60* and the third is at 120*, it will produce the small arc.
If the second point is at 120* and the third is at 60*, it will produce the large arc.

How is it supposed to work?  Should it always draw the small arc?

DEL-TRONIK, 2023-11-15:
The expected result should be like this:

[image]

If you open the previous file with a version 34387 (or older) of freecad this will be displayed correctly, but with more recent versions, the arc is drawn in reverse, I don't know why this behavior.

However, if I reproduce the example as illustrated in the previous video, it works even with newer versions of freecad. Why does it change the location of the arc with new versions if I simply open my previous file (create with the version 34387 of freecad) with the new version?

WandererFan, 2023-11-15:
If I look back at older versions of the arc creation code, it defines the arc as running from "end angle" to "start angle" .   The current code makes the arc from "start angle" to "end angle".  

The old code does not set the "reversed" property of the arc properly.   If the arc runs from a larger angle to a smaller angle, the "reversed" property must be set to true.   

If the arc in your file runs from large angle to small angle and the reversed property is not correct, the current code will draw the larger part of the circle instead of the smaller part.

DEL-TRONIK, 2023-11-22:
Okay, so there has been a change in the code, which explains this behavior. If I redo the previous arc step with the new version of FreeCAD and open the file with an older version, the arc is reversed too. So there's the same problem (between old and new versions) in reverse, which confirms all this. So, what's the solution? I have a lot of LED files of this type and I'd hate to have to rework them all to modify the arc on each one. I'm going to keep an old version of freecad for these copies to keep them up to date. Thanks.

luzpaz, 2024-07-29:
Is this issue still relevant, @WandererFan  ?

luzpaz, 2024-10-12:
@WandererFan :point_up_2:

WandererFan, 2024-10-19:
I would say that this one can be closed.   The current version might behave differently than earlier versions, but the earlier version were wrong and not it works correctly.

closed 2024-10-19 by luzpaz