FreeCAD/FreeCAD #11394: [Problem] C++ DXF importer wrongly prioritizes the MEASUREMENT variable

Roy-043, 2023-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The C++ DXF importer wrongly prioritizes the MEASUREMENT variable. It should first check INSUNITS. Only if this variable is missing or set to zero, should MEASUREMENT be considered.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=82546

Test file:
[DXF_import_test_cpp.zip](https://github.com/FreeCAD/FreeCAD/files/13348195/DXF_import_test_cpp.zip)

For reference: algorithm used by the legacy importer:
https://github.com/FreeCAD/FreeCAD/blob/8b6d133c26857363bedec2ba31bacbecb2a9bf89/src/Mod/Draft/importDXF.py#L2079-L2092

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35024 (Git)
Build type: Release
Branch: main
Hash: 4ff00ef99869e35295a5a17b8f486901504c655b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-14:
Hey @kpemartin would you be open to taking a look at this ?

yorikvanhavre, 2023-11-16:
Probably the problem is at https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Import/App/dxf/dxf.cpp#L3920
MEASUREMENT should be considered only if INSUNITS is zero... Probably we need to add something like `if (m_eUnits == eUnspecified) {`

tmksw, 2023-12-10:
This issue is related to #6461 
[image]

tmksw, 2023-12-10:
I have create a PR which should resolve this specific issue however it would be good to have some unit test in place to cover this and similar scenarios.

closed 2023-12-14 by yorikvanhavre