FreeCAD/FreeCAD #11397: [Assembly] UX: Lack of feedback when trying to add links to an unsaved assembly

iromero91, 2023-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to add an external link to an assembly without saving it first, it will appear to silently fail.

Steps to reproduce:
* Have the report view hidden (with the new shiny non-intrusive notifications)
* Create a new document and an assembly, do not save
* Press the Insert Link button in the toolbar, the task panel will open
* Select a part from an external file and hit OK
* The task panel will close with nothing have happened and no error reported to the user

Note: if you then manually enable the report view you can see that the WB did print out the fact that the assembly needs to be saved in order to use external links, but this is not visible when using non-intrusive notifications

### Full version info

```shell
OS: Ubuntu Core 22 (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34988 (Git) Snap 832
Build type: Release
Branch: main
Hash: dfcfdad0400d5635cc03fcaa4849a2246bbd3e14
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.17.1
  * sheetmetal 0.2.63
  * LCInterlocking
  * fasteners 0.4.62
```
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I suggest at least elevating the notification to a warning or error so that it will appear when the report view is hidden. Perhaps some user friendlier way to remind the user to save could be implemented, perhaps including a prompt to save.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-12-08:
Thanks for the bug report !
It was actually not silent, there was a python print. But this is not enough indeed.
So instead I have changed it so that when you try to insert a part that is not from current document, then a qmessagebox will prompt you to save the document before inserting it.

iromero91, 2023-12-12:
> Thanks for the bug report ! It was actually not silent, there was a python print. But this is not enough indeed. So instead I have changed it so that when you try to insert a part that is not from current document, then a qmessagebox will prompt you to save the document before inserting it.

Thanks for the fix, yeah it wasn't technically silent, said as much in the original bug description. Though from a non-developer using the default quiet notification system, it is effectively silent!

PR #10764 (2024-02-13): [Assembly] Solve the assembly (Card 9).
Adds AssemblyObject and viewer in cpp that handle the solving by connecting to ondsel solver.
Fixes #10753
Fixes #10749
Fixes #10754
Fixes #10479
Fixes #11311
Fixes #10751
Fixes #11397
Fixes #10815
Fixes #11606
Fixes #10752
closed 2024-02-13 by chennes