FreeCAD/FreeCAD #11404: [Problem] Part_cut & Part_common commands swapped on 0.21.1

RocaroDev93, 2023-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On the tool bar in **Part** mode, the command `cut` execute a `common` command instead of a `cut` common. And the `common` button executes a `cut`command instead of a `common` command.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/Belgium (fr_BE)
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-15:
I cannot confirm this.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

luzpaz, 2023-11-15:
@RocaroDev93 can you show a screencast of this ?

closed 2023-11-16 by Roy-043

RocaroDev93, 2023-11-16:
@luzpaz I tried to reproduce the bug but I haven't reached the same result. But I can share with you the project file where the error occurs.

luzpaz, 2023-11-16:
@RocaroDev93 zip it (make sure the file extension is .zip) and drag drop in to github textbox here

RocaroDev93, 2023-11-16:
Here is the project file

[common_cut_commands_swapped.zip](https://github.com/FreeCAD/FreeCAD/files/13377945/common_cut_commands_swapped.zip)

Roy-043, 2023-11-16:
The "Revolve001 (Solid)" object has flipped normals. It is inside-out. You can verify this by changing its Lighting property to "One side" (on the View tab of the Property editor). With default settings it will then become completely black. You can fix this with the [Part_ReverseShape](https://wiki.freecad.org/Part_ReverseShapes) command.

RocaroDev93, 2023-11-17:
I think you should add an other way to display that the shapes are reversed because it's easy to make the mistake and pretty hidden to know that information

Roy-043, 2023-11-17:
This issue is actually pretty rare. How the faulty object came about I do not know. The file does not contain its parametric history.

luzpaz, 2023-11-17:
How would one represent a inside-out flipped normal in the UI, just out of curiosity?

FlachyJoe, 2023-11-17:
A check-and-fix for flipped solid exists at https://github.com/FreeCAD/FreeCAD/blob/f741d28b43b93a08ac2f3700e0df2e831b873576/src/Mod/PartDesign/App/FeatureHelix.cpp#L300-L303

Should we consider to use it everywhere?

RocaroDev93, 2023-11-17:
@FlachyJoe I don't think there is any fix to do. It was an error I made during the design. I probably applied a reserveShape on my object at a moment.

@luzpaz I don't know what solution would be the best in this case to inform users that an object has reversed shape. You could add the parameter in the object parameters list. I checked and I have found it. Using a different color for inside side of a shape

luzpaz, 2023-11-17:
@FlachyJoe that is cool. I wonder if there is a way to softly present the option to fix the flip instead of auto-flip for the user ?

FlachyJoe, 2023-11-17:
@luzpaz does exist a use case for reversed solid? Isn't Part_ReverseShape only created to fix broken results?