FreeCAD/FreeCAD #11408: C++ DXF importer fails if text contains q2uotation marks

kpemartin, 2023-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If a DXF file contains text which includes a double-quote (") the C++ DXF importer will fail with any of various errors depending on the remaining contents of the text.

The reason is that the code creates the text object by running generated Python code in the interpreter, and fails to properly escape the contents of the string.

Other string content which is significant to the Python string literal syntax would also be misinterpreted.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35033 (Git)
Build type: Release
Branch: Issue11394
Hash: 8b6d133c26857363bedec2ba31bacbecb2a9bf89
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
```
```

### Subproject(s) affected?

None

### Anything else?

The error text in the console includes:
<class 'RuntimeError'>: ('invalid syntax', ('<string>', 1, 55, 'Draft.make_text("This text contains a double-quote: " which will cause its import to fail",pl, height=0.881944)\n'))

Note how the code to be interpreted ('Draft.make_text...' contains a string literal closed by the quote after the text 'quote: ' and the text 'which will cause' is being interpreted as more of the arguments to make_text.

The code also fails to apply any of the other properties of the text such as color, 3D orientation, etc. although some of these might be applied elsewhere.

I have a sample file in a zip file but have been unable to attach it here ("Something went really wrong, and we can't process that file")

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-15:
> I have a sample file in a zip file but have been unable to attach it here ("Something went really wrong, and we can't process that file")

@kpemartin can you put it on a dropbox ?

yorikvanhavre, 2023-11-16:
Looks to me we need to escape double quotes (maybe singles too) in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Import/App/dxf/ImpExpDxf.cpp#L333

kpemartin, 2023-11-16:
...and backslashes, and whatever else.
I have a fix in the works that avoids the problem by calling the Python methods directly rather than generating Python code an interpreting it.
Even that seems awkward so I'll also see if I can use direct C++ calls.

yorikvanhavre, 2023-11-16:
Yes, the C++ calling python is weird, but it's mainly because the best text object we have ATM is a Draft (python) object, so that leaves little options...

PR #11458 (2023-11-20): Use PyObject_CallObject rather than interpreting generated code
Rather than generating Python code and interpreting it, which, in C++, is fraught with issues of creating the correct syntax for a str token when the text contains certain special characters like double-quote and backslash, the modified code makes call(s) to PyObject_CallObject which takes the C++ string and makes the appropriate conversion itself. The steps in building the Position passed to make_text are also done using c++ objects until the final Python Placement object is needed.

Closes #11408
closed 2023-11-20 by yorikvanhavre