FreeCAD/FreeCAD #11410: [WB Path] Millface "Clear Edges" does not respect walls higher than selected face

blazini36, 2023-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This has been a thing as long as I can remember, just want to call it out so it eventually gets addressed.

"Clear edges" was added at some point to extend the operation beyond the boundbox. This is good but it is generally not of any use because it will push into a vertical face that is higher than the horizontal face the operation is working on.

Without "Clear Edges" selected  the toolpath will leave remnants on the face that was milled down:
[image]

With Clear Edges selected it extends beyond the edges of the face which is ideal but it does not respect the wall:
[image]

Adding the wall to base geometry does not make a difference.

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35039 (Git)
Build type: None
Branch: main
Hash: e937cc1efc3e210655575a408425098fafef39e7
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * freecad.gears 1.0.0
  * kicadStepUpMod 10.14.5
  * dxf-library
  * fasteners 0.4.54
  * Assembly4 0.12.6
```
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-06-18:
I believe the desired effect can be achieved with current functionality.

I roughly recreated the example model.  Then created a pocket operation selecting the top face and the bottom face of the pocket.
Then enabled extensions and selected the outer wire of the top face.
This creates a pocket that goes from the stock top to the top of the first face.  It avoids the vertical wall and extends past the edge of the shelf on all other faces.

Unless I'm missing something, I will close the issue.

[image]

luzpaz, 2025-01-28:
@blazini36 👆

sliptonic, 2025-05-19:
Closing as resolved

closed 2025-05-19 by sliptonic