FreeCAD/FreeCAD #11411: [WB Path] MillFace does run final pass in any case

blazini36, 2023-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is another one that's been present in the past few versions:

When millface is set with a "Finish Step Down" of 0, the expected behavior is that it does not make a special shallow last path. If "Finish Step Down" is selected it is expected that the last pass just makes room for the desired value to make the last pass. In both cases the final pass is the "Final Depth"

What actually happens these days is with "Finish Step Down" unselected, the last pass is calculated 1 "step down" height above the face. I.E. if "Step Down" is 0.07in, then the last pass will be 0.07in above the face, meaning it skips the whole pass. 
[image]

If "Finish Step Down" Is applied then the last pass will be that height above the face. I.E. if it is set to 0.010in then the last 0.010in pass will never happen. 

In it's current form, the only way possible to get a final pass at the intended depth is to fudge it by dropping "Final Depth" lower than the intended depth, then setting "Finish Step Down" to the fudged amount. For Example:

Intended face final depth: 0.512
Set "Final Depth": 0.511
Set "Finish Step Down": 0.001

I'm sure this isn't the intended behavior, it's not how it worked in the past.

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35039 (Git)
Build type: None
Branch: main
Hash: e937cc1efc3e210655575a408425098fafef39e7
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * freecad.gears 1.0.0
  * kicadStepUpMod 10.14.5
  * dxf-library
  * fasteners 0.4.54
  * Assembly4 0.12.6
```
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2023-12-03:
This bug did not exist in 0.20.2 and did exist in 0.21.0.

[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * Assembly4 0.50.4
  * fcgear 1.0.0
  * Plot 2022.4.17
  * DynamicData 2.53.0
  * BIM 2021.12.0
  * parts_library
[/code]
[NoFinishPass.zip](https://github.com/FreeCAD/FreeCAD/files/13539401/NoFinishPass.zip)

bgbsww, 2023-12-07:
May I be assigned, please?

closed 2023-12-11 by sliptonic

luzpaz, 2023-12-12:
Thanks for all your awesome efforts in various workbenches of FreeCAD :pray:

PR #11673 (2023-12-11): Tweak section tolerance to make base layer of paths work again
Fix #11411 

I've spent a while going through all the Area and AreaParams code, and most of Path.  As far as I can see, there are several other places using precision at 1e-06.

From a practical milling standpoint, 0.1 microns or 1e-04 mm or 4e-06 inches appears to be the limit, so this seems reasonable.

I wasn't able to find any other ways to fix this problem, nor was I successful in pinpointing what change between 0.20.2 and 0.21.0 triggered it.