FreeCAD/FreeCAD #11412: [Feature request] More flexible circle to circle and circle to line dimensions

pierreporte, 2023-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, circle to circle and circle to line distances consider the smallest distance between the two entities. It is often needed to dimension the largest distance. There can be more complex situations like dimensioning from the closest point on one side to the farthest on the other side.

FreeCAD could understand where the users clicks on the circles to determine the right dimension. For example, if the farthest side is clicked, then FreeCAD could show the largest dimensions. A dialog could be a good option too, like Alibre (see below).

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git) AppImage
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

These are Alibre screenshot. It is not very clever so it shows a dialog with all the possibilities represented graphically. The user chooses the right one in the list.

![image](https://github.com/FreeCAD/FreeCAD/assets/2029853/b48dc447-39e5-4794-9170-a20a13b3738b)

![image](https://github.com/FreeCAD/FreeCAD/assets/2029853/cf24e222-ddcf-4c9d-aa0e-d0a30a2d8987)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dgrabulosa, 2023-11-18:
FreeCad forum discusión
https://forum.freecad.org/viewtopic.php?t=80886

maxwxyz, 2024-02-16:
I'll close this in favor as duplicate of #11918

maxwxyz, 2024-02-16:
I'll close this in favor of
Duplicate of #11918

closed 2024-02-16 by maxwxyz