FreeCAD/FreeCAD #11413: [Feature request] Sketcher: add a context menu for constraints

pierreporte, 2023-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is about constraints like coincidence, tangency, parallelism, etc. and **not** about dimensions.

Currently, to add constraints to sketch geometry, it is necessary to select a command in the toolbar, then click on the geometry to constrain. There is also a few of them available with pressing the `M` key with the new `Sketcher_Dimension` tool. There could be a more complete and more visual way, using `Sketcher_Dimension`, but not the `M` key. Besides, if this feature request is implemented, the `M` key could be made for dimensions only and not constrains.

What I propose is a context menu activated by a right click when `Sketcher_Dimension` is active. It would show the constrains that could by applied to the selection. For example, selecting a point and a line would propose a coincidence constraint. Two circle would allow to make them tangent, concentric, or equal. You could make your lines perpendicular, parellel, tangent, etc. Only exception for two lines: to avoid a duplicate action, it could propose to set a distance between them being parallel and not just a parallel constraint. Because the right-click is currently used to exit a sketch tool, the first item in the context menu should be a cancel entry (you could cancel with `Esc` as well, which isn’t possible right now).

This new feature exists in Catia and it is very practical. You can fully-constrain complex sketches with a single tool. Catia basically hides all constraint tools because of this. Catia uses this context menu to also make the dimension (primary use of `Sketcher_Dimension`) a reference dimension.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git) AppImage
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
```

### Subproject(s) affected?

Sketcher

### Anything else?

Catia screenshot:

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-17:
The right click is used to cancel the tool in sketcher. So right click may not be ideal. But perhaps the M key could open a pie-menu with the different modes available.

pierreporte, 2023-11-17:
@PaddleStroke yes, it’s not necessary to be a right click. The `M` key is not very discoverable as you can’t really know what will be proposed, so adding a menu would be great anyway. The pie menu is a good idea if it uses the icons in the toolbar.

PaddleStroke, 2023-11-24:
Note : the video you posted in #11414 (https://www.youtube.com/watch?v=I3HUKB1np3E) show the solidworks way of doing it. It seems rather clean way of doing it (without right mouse click)

pierreporte, 2023-11-24:
@PaddleStroke Speaking from experience, the Solidworks way is clean indeed but slower and less efficient than the Catia way. Actually one of the few thing I love in Catia. In any case it will be a great addition to FreeCAD.

maxwxyz, 2024-01-01:
What if the first entry in the right click menu is "exit tool" or "abort"? That is the way how it is handled in KiCAD for example. Regardless of the right click you could further exit the tool with *esc*

maxwxyz, 2024-01-07:
@pierreporte https://github.com/FreeCAD/FreeCAD/pull/11884 could help

pierreporte, 2024-01-08:
@maxwxyz You may be right. Constraints would be visible when the dimension tool is active and geometry is selected. The constraints have to be filtered to what is possible to do depending on the selection. Although the “select a symmetry axis” feature may be a little harder to implement.

pierreporte, 2024-01-28:
PR #12161 implements part of this feature request: provide a context menu for already selected geometry.

maxwxyz, 2024-02-16:
Now you can also directly right click on geometry to pull up the constraints

pierreporte, 2024-04-22:
@maxwxyz for the UI/UX project.

luisDias89, 2024-06-04:
@pierreporte
@PaddleStroke 
 I disagree with the right-click button being used to open the menu; it should be automatic. After selecting lines, the menu should appear after half a second or a second, and if the mouse doesn’t move, the menu should show up automatically. If the cursor moves toward the menu, it becomes active; otherwise, it disappears. (Like solidworks)

PS: the right button is for properties of object. and not for smart tips.
      and M key need to a training with tutorial, its not intuitive

This menu should be applied to all workbenches, with each workbench having its intelligence to provide tips on features that can be used based on the current user selection. Initially, it could be implemented in the Assembler,Sketcher,partdesign, and then expanded to other workbenches.
Each workbench has its own specifications

luisDias89, 2024-06-04:
In terms of programming logic, there should be a structure that detects selections and counts the time to trigger. After the trigger, it enters a function that detects which workbench is currently active and corrects the "smart tip" function of that workbench. If it doesn't exist yet, it does nothing. If it does exist, it passes what was selected to be processed inside the function. The function now has the responsibility of infixing Features with what was selected that can be used on the selection.

maxwxyz, 2024-06-04:
Feel free to create a PR.

luisDias89, 2024-06-04:
See how it works starting at minute 17:30
https://www.youtube.com/watch?v=_nyLnRaP0Yc&list=PLm1W92pCWeKNo5PqIIi7fn51-tWvzpPOv

pierreporte, 2024-06-04:
@luisDias89 You misunderstand the intent of this feature request. The goal is to mimic a Catia feature. You want something more general. They are two different things.Both are useful.

kaiwas, 2024-06-11:
> The `M` key

Is it possible to assign some other more convenient button wherever “M” is currently used?
This key is located in an inconvenient place. It is very difficult to “record” it into muscle memory and most often you have to look for it with your eyes.
It's also located in slightly different places depending on the keyboard model (a few millimeters of displacement can be difficult for muscle memory).
Maybe the TAB key could be suitable as a replacement for switching modes wherever it is necessary to change the mode. This key is everywhere in one place. It is quite large and you don’t need to look with your eyes where it is located.
Should I create a feature request on Github or is such a change basically impossible?

Mannshoch, 2024-08-30:
Mokup for FreeCAD: https://forum.freecad.org/viewtopic.php?p=582953#p582953

I would vote for. Unify and simplify (2D and 3D) of mate, relations and measures into one set of tools.
As an example, a parallel mate could be enhanced with a value in mm and it changes to a dimension, enhance it with an angel instead and it will change to an angel dimension. And vice versa. Remove a the distant value in the settings and it convert to a parallel constraint. change the value to 0 then it changes to a tangent.

pierreporte, 2024-08-30:
@Mannshoch The issue is more about how the menu is displayed to the user, and less about how it looks. Though this context menu is common in CAD applications when sketch geometry is selected, which is a good thing and is worth to implement, what I propose here is to also activate it when using the generic constrain tool like in Catia. It is one of the few UX perks of Catia.