FreeCAD/FreeCAD #11414: [Feature request] Sketcher: add more automatic constraints (with helpers)

pierreporte, 2023-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, the sketcher automatically adds the following constraints depending on the context: coincidence, tangency, vertical, and horizontal. There should be more like perpendicular, parallel, equal, or even midpoint (see #10704).

In other software, this works using the local context. If you approach a line with another line, it will propose you to make it perpendicular. Going alongside a line will show you that it can be parallel. This is done using helpers. Helpers can be activated with actions: for example a perpendicular constraint only if you hover the desired line.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git) AppImage
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
```

### Subproject(s) affected?

Sketcher

### Anything else?

On the following video, you can see the yellow dashed lines that will transform into constraints in you follow them. The presenter doesn’t use the full power of the feature though, as he could have at least made his arcs tangent with the lines.

https://www.youtube.com/watch?v=I3HUKB1np3E

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2023-12-14:
#6240 is related because auto-constraints are a subset of what this issue is about.

pierreporte, 2024-01-07:
The following article gives details on how Catia handles this, with pictures clearer than the video: 

http://visualtutor.blogspot.com/2009/07/sketcher-smartpick.html

[(Archive link)](http://web.archive.org/web/20240107164456/http://visualtutor.blogspot.com/2009/07/sketcher-smartpick.html)

pierreporte, 2024-01-09:
@PaddleStroke what do you think of this?

PaddleStroke, 2024-01-09:
I think it should be implemented. I have not done so yet because I don't
know auto constraint code. Also I have plenty on my to-do. I'll probably do
it at some point if no one else has done it.

maxwxyz, 2024-02-25:
Related: https://github.com/FreeCAD/FreeCAD/issues/5586

pierreporte, 2026-05-29:
@PaddleStroke yes indeed! It just lacks the hints. See these Solidworks videos that shows a colinear hint line and a perpendicular line hint, that also covers tagent and its normal. In the second video you can see that if you hover a different lines, then the hint lines are perpendicular/parallel to that one. In Solidwosk they are called inference lines.

https://www.youtube.com/shorts/Is88RIJj6eA

https://www.youtube.com/watch?v=V8lYpxqzkNo