FreeCAD/FreeCAD #11424: The Sketcher B-spline visuals menu doesn't indicate what's on or off

Stokestack, 2023-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I imported some shapes as SVGs and converted them to sketches. Upon opening them in sketcher, the canvas was littered with a mass of green data about the sketch, and unusable.

A Web search revealed a forum post that pointed me to the **_Sketch/Sketcher visual/Show/hide B-spline information layer_** menu, but this menu is dysfunctional because it doesn't indicate what's on or off. Thus there's no way to tell what's cluttering up the screen, and no way to tell what you've already toggled. You have to sequentially go through it and remember to try the first, then the second, then the third, etc.

A related problem is that the text of every entry starts with "show/hide." When every line starts with the same string, it is not useful for identification and should be removed. For view options, the state of each one is typically indicated with a checkbox at the beginning of the line.

[image]

### Full version info

```shell
[code]
OS: macOS 14.1.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34651 (Git)
Build type: Release
Branch: main
Hash: aa5bea4e2e8089bf00f61075a0860dacda42dc8b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-17:
@AjinkyaDahale would you have a moment to look into this ?

Stokestack, 2023-11-17:
I think I have a screen cap of a menu using checkboxes somewhere if you guys would like an example. There may be similar menu-status ambiguity elsewhere in FreeCAD, but I figured it's best to address specific areas at a time.

AjinkyaDahale, 2023-11-18:
Best handled by someone already doing UI. Is there presently a case within freecad where this happens?

Off the top of my head, I cant think of a neat way to have both the check mark _and_ the icon logo. Additionally, these menu options also have equivalent toolbar buttons. How would it look there?

Stokestack, 2023-11-18:
There are toolbar buttons for all of these? I haven't seen those. Do they also not indicate their state? Did you find the example screen shot of check + icon in a menu unacceptable? The other option is a "selected" and "unselected" state for each icon. The complexity of these icons does make that a bit more difficult than some, because you can't just draw an outline around an active one... it would obscure a good portion of the diagram.

However, you could have versions of each with a "selected" background color. The challenge there is finding a color that will be visible regardless of the system's color scheme.

AjinkyaDahale, 2023-11-28:
> There are toolbar buttons for all of these? I haven't seen those. Do they also not indicate their state?

When in sketcher edit mode you can see. No indication though.
[image]

> Did you find the example screen shot of check + icon in a menu unacceptable? 

I would need to look into Qt to see if that is possible. Other than that it's just a little busy (something we can live with).

> The other option is a "selected" and "unselected" version of each icon. The complexity of these icons does make that a bit more difficult than some, because you can't just draw an outline around an active one... it would obscure a good portion of the diagram.

"Unselected" can be confused with "Disabled". I'd go for checkmark for sure.

> However, you could have versions of each with a "selected" background color. The challenge there is finding a color that will be visible regardless of the system's color scheme.

Not to mention there are multiple themes available.