FreeCAD/FreeCAD #11429: [Feature request] Part Design: double-click & context menu to edit

pierreporte, 2023-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Part Design Workbench, to edit a sketch or a feature, it is necessary to find it in the tree view, then double click on it, or right-click and select the option in the context menu. This isn’t practical in a complex model because:

1. it may be difficult to locate precisely the feature or sketch;
2. it is not highlighted when it is selected in the 3D view;

I propose to switch to edit mode when the user double-clicks on the feature or sketch directly in the 3D view (currently, it selects the last feature of the body.) I also propose that a right-click gives the option to edit what is right-clicked on. This menu could have an option to select the body in order to keep the current behavior close to where it is now.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git) AppImage
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-17:
I think this may be a duplicate.

pierreporte, 2023-11-17:
@luzpaz I haven’t found anything in the issues or the forum. What issue are you thinking about?

M4xTheMan, 2023-11-28:
I'm not sure I understood you correctly. Do you want FreeCAD to figure out which feature in the feature tree you'd like to edit instead of the last one and the input is a click in the 3D-view on a face for example?

Simple example: A model consisting of a Pad, a Revolution and a Groove (from bottom to top).
[image]

Which feature should be selected in the tree view, when...

I mark Face14 at the top:
[image]

I mark Face9 on top of Pad:
[image]

?

Here is my example file (changed the ending from FCStd to zip):
[example_issue_11429.zip](https://github.com/FreeCAD/FreeCAD/files/13493154/example_issue_11429.zip)

pierreporte, 2023-11-28:
Yes, I want FreeCAD to understand that:

1. if I double-click a sketch, I want to edit that sketch;
2. if I double click somewhere on a part, I want to edit the feature that generated what was clicked on.

In your example, I think that Face14 and Face9 are generated by the revolution. Double-clicking on the conical face would edit the groove feature. Double-clicking on the rectangular faces would edit the pad. Double-clicking on the visible sketch would edit that one.

M4xTheMan, 2023-11-28:
One could argue that Face14 was initially created by the Revolution:
[image]

If it would be possible at all, I assume that the TNP ([Topological_naming_problem](https://wiki.freecad.org/Topological_naming_problem)) stuff has to get merged before hand.

pierreporte, 2024-01-17:
@maxwxyz Are you able to do something with your new contextual right-click menu?

maxwxyz, 2024-02-18:
@pierreporte this was just done in the Sketcher WB but I guess something like this could and should be done for other WB as well.

maxwxyz, 2024-03-20:
In the RT link branch the selection in the 3D view (geometry) selects the Part Design feature that created the geometry. Guess we need TNP solved first.

pierreporte, 2024-04-22:
@maxwxyz for the UI/UX project.

maxwxyz, 2024-08-30:
When https://github.com/FreeCAD/FreeCAD/issues/16164 is implemented, a double click on an element (e.g. a face) in the 3D view should open the edit task dialog of the feature which created it.

Example: Double clicking on the face (fillet) should open the dialog to edit the fillet feature which created this fillet.
[image]

Similar issue for sketches: Double click a sketch in the 3D view should open the sketch:

maxwxyz, 2024-08-31:
in the RT link branch this is somewhat implemented: hovering the feature in the tree view, displays the face/geometry which was created by that feature. It does not work for double click to edit and not the other way around (select in 3D view and display the tree view feature).

https://github.com/user-attachments/assets/b1322f48-b8f4-4d0c-9468-63d42c5455eb