FreeCAD/FreeCAD #11434: [Problem] Sketcher_ConstrainBlock does not work for points

pierreporte, 2023-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce:

* Select a point, then `Sketcher_ConstrainBlock`: FreeCAD displays an error message saying that the selection is invalid.
* Select `Sketcher_ConstrainBlock` then try to click on a point: the mouse pointer indicates an impossible action and the status bar says that the selection isn’t allowed.

Everything but points can be blocked. For consistency, it should be possible.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35046 (Git) AppImage
Build type: Release
Branch: main
Hash: 6a66034ce27290e7b385a933d77375f55492d194
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2023-12-19:
This is fixed in the latest weekly (version below) but now it only works for points and not for everything else!

```
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git) AppImage
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * OpenDark 2023.12.1
```

pierreporte, 2024-01-01:
Reopening because I messed up with `ConstrainBlock` and `ConstrainLock`.

maxwxyz, 2024-01-05:
Not a bug in my opinion, currently points are not supported by this constraint. Should be a feature request if this is really necessary

pierreporte, 2024-01-05:
@maxwxyz I tend to agree with you. The actual issue is that there are two different constraints. It’s a bit like coincidence and point on object. They do the same thing but one works only for points and the other for everything else. You just need to know which one to use.

@PaddleStroke Do you thing merging `Sketcher_ConstrainBlock` and `Sketcher_ConstrainLock` would be a good idea, like you did for the new coincidence tool?

Should I close this issue and #11778 to create a new one for the merge?

maxwxyz, 2024-01-05:
They actually don't do the same. **Block** just blocks it, whereas **lock** creates two distances (horizontal and vertical) as one way to constrain them. You can edit the dimension afterwards.

pierreporte, 2024-01-05:
I know but the way `Sketcher_ConstrainLock` does not makes sense in regard to its name. The dimension tools are there to define actual distances. Why was `Sketcher_ConstrainLock` designed this way?

maxwxyz, 2024-01-05:
I'm familiar with the history but I guess the lock is there to easily make a point fully constrained (and fast). Maybe lock could be entirely removed with the new dimension tool.

pierreporte, 2024-01-05:
If `Sketcher_ConstrainLock` is removed, there would be no way to block a point in position so it would be necessary to extend `Sketcher_ConstrainBlock`. It would be necessary if non-construction points are implemented, as I suggested in #10100 and #6884.

PaddleStroke, 2024-01-06:
Imo Sketcher_ConstrainBlock should indeed support points. This is a feature request.
Regarding the Lock constraint : initially sketcher_dimension was doing a lock constraint on point selection. But people thought it was overkill and asked to remove it as the primary mode, so now you need to hit M key to go to lock.
So overall I think lock tool is not very usefull with the sketcher_dimension. However it does not clutter much in the sketcher_dimension dropdown so I would not remove it either.

I think the name 'lock' is perhaps not ideal. I prefer the Catia concept of lock constraint which we don't have : You select a bunch of geometries and lock them together. It's very usefull for imported DXF (say you import the FCC logo for instance). Then the imported shape is not constrained and constraining it can be very hard. In catia you can lock the group together. Then the whole group can move, but as a single entity.

pierreporte, 2024-01-06:
Yes, a tool to make a block of geometry is definitely needed. It is not exclusive to Catia though.

Having both `Sketcher_ConstrainLock` and `Sketcher_ConstrainBlock` is misleading even though the first does not add clutter. In fact I created this issue because I thought That `Sketcher_ConstrainBlock` changed behavior and that I made a spelling mistake.

The current `Sketcher_ConstrainLock` behaviour could be partially emulated with tool settings for points.

Anyway, how often do you need to lock a point in place with setting both dimensions? In my career as a mechanical engineer, it basically never happen. One dimension yes but it’s more often constrained using sketch geometry (or external geometry) or just locked in place in situations like an imported DXF.

maxwxyz, 2024-03-30:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/5844

PaddleStroke, 2025-01-28:
Still relevant

pierreporte, 2025-01-28:
@PaddleStroke this could be fixed by merging the two commands into one, like it was done for coincident and point on object. They are strictly complementary. Of course an option to give back the separate tools would be welcome for the same reason it was done before.

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/74a6360d044272b9bfed637ded7709a2d5d84eab

Block accepts point geometry too, both from a selection and while picking. Vertices belonging to an edge are still rejected.