FreeCAD/FreeCAD #11437: [Problem] CMake Error at src/3rdParty/CMakeLists.txt:14 Configuring incomplete, errors occurred!

superbonaci, 2023-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In-source building

```
$ cd freecad-source
$ cmake . -DPYTHON_EXECUTABLE=/usr/bin/python3
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
CMake Error at src/3rdParty/CMakeLists.txt:11 (message):
  The OndselSolver git submodule is not available.  Please run

          git submodule update --init

CMake Error at src/3rdParty/CMakeLists.txt:14 (add_subdirectory):
  The source directory

    /builds/freecad-source/src/3rdParty/OndselSolver

  does not contain a CMakeLists.txt file.

etc.

=================================================
Now run 'cmake --build /builds/freecad-source' to build FreeCAD
=================================================

-- Configuring incomplete, errors occurred!
```

### Full version info

```shell
source build
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

superbonaci, 2023-11-18:
To fix it, this command has to be run before cmake:
git submodule update --init

Which is not included here:
https://wiki.freecad.org/Compile_on_Linux

Then you can run `make -j$(nproc --ignore=2)` and works fine.

howie-j, 2023-11-18:
The wiki have been updated today by Hasecilu to also clone the OndselSolver submodule when cloning freecad:

`git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD.git freecad-source`

`git clone --recursive` means there is no need to run `git submodule update --init` to clone the submodule.

closed 2023-11-18 by chennes