FreeCAD/FreeCAD #11438: [Problem] Compile on Arch linux fails

superbonaci, 2023-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Compile for Arch:
```
$ make -j$(nproc --ignore=2)
/home/freecad-source/src/3rdParty/OndselSolver/OndselSolver/Array.h: In instantiation of void MbD::Array<T>::zeroSelf() [with T = double]:
/home/freecad-source/src/3rdParty/OndselSolver/OndselSolver/Array.h:73:14:   required from here
/home/freecad-source/src/3rdParty/OndselSolver/OndselSolver/Array.h:75:35: warning: comparison of integer expressions of different signedness: int and std::vector<double, std::allocator<double> >::size_type {aka long unsigned int} [-Wsign-compare]
   75 |                 for (int i = 0; i < this->size(); i++) {
      |                                 ~~^~~~~~~~~~~~~~
[ 40%] Linking CXX shared library libOndselSolver.so
[ 40%] Built target OndselSolver
make: *** [Makefile:136: all] Error 2
```

### Full version info

```shell
source code
```

### Subproject(s) affected?

Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-18:
heads up @sliptonic 

@superbonaci Please make sure to explicitly say Arch Linux since Arch in FreeCAD means the Architecture workbench. We also need to see the cmake report of dependencies to know with what dependencies you compiled with.

superbonaci, 2023-11-18:
Does hit help?
```
$ cmake . -DPYTHON_EXECUTABLE=/usr/bin/python3
-- Install configuration: ""
-- Compiler: GNU, version: 13.2.1
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.27.8
-- Compiling with Qt 5
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
=====================================================
Doxygen not found, will not build documentation.     
=====================================================

-- Checking for connection to GitHub...
-- GitHub connection established for FetchContent
-- Module support is disabled.
-- Version: 9.1.0
-- Build type: 
-- CXX_STANDARD: 17
-- Required features: cxx_variadic_templates
-- fmt was downloaded using FetchContent into /soft/freecad-source/_deps/fmt-src
-- PyCXX found:
--   Headers:  /soft/freecad-source/src
--   Sources:  /soft/freecad-source/src/CXX
--   Version:  7.1.7
-- -- Found OCE/OpenCASCADE version: 7.7.2
-- -- OCE/OpenCASCADE include directory: /usr/include/opencascade
-- -- OCE/OpenCASCADE shared libraries directory: /usr/lib
-- Found OpenGL: /usr/lib/libOpenGL.so   
CMake Deprecation Warning at /usr/lib/cmake/vtk/vtk-config.cmake:79 (message):
  The new name for the 'vtkCommonCore' component is 'CommonCore'
Call Stack (most recent call first):
  cMake/FreeCAD_Helpers/SetupSalomeSMESH.cmake:29 (find_package)
  CMakeLists.txt:69 (SetupSalomeSMESH)

-- Found MPI: TRUE (found version "3.1")  
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
CMake Warning at /usr/lib/cmake/vtk/VTK-vtk-module-find-packages.cmake:3130 (find_package):
  By not providing "FindFastFloat.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "FastFloat", but CMake did not find one.

  Could not find a package configuration file provided by "FastFloat"
  (requested version 5.1.0) with any of the following names:

    FastFloatConfig.cmake
    fastfloat-config.cmake

  Add the installation prefix of "FastFloat" to CMAKE_PREFIX_PATH or set
  "FastFloat_DIR" to a directory containing one of the above files.  If
  "FastFloat" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /usr/lib/cmake/vtk/vtk-config.cmake:159 (include)
  cMake/FreeCAD_Helpers/SetupSalomeSMESH.cmake:29 (find_package)
  CMakeLists.txt:69 (SetupSalomeSMESH)

-- Could NOT find utf8cpp (missing: utf8cpp_INCLUDE_DIR) 
-- VTK components: CommonCore;CommonDataModel;FiltersVerdict;IOXML;FiltersCore;FiltersGeneral;IOLegacy;FiltersExtraction;FiltersSources;FiltersGeometry;IOMPIParallel;ParallelMPI;hdf5;FiltersParallelDIY2;RenderingCore;InteractionStyle;RenderingFreeType;RenderingOpenGL2
-- Found MPI: TRUE (found version "3.1") found components: C 
-- Found OpenGL: /usr/lib/libOpenGL.so  found components: OpenGL 
-- Could NOT find freetype (missing: freetype_DIR)
CMake Warning at /usr/lib/cmake/vtk/VTK-vtk-module-find-packages.cmake:2859 (find_package):
  By not providing "Findfmt.cmake" in CMAKE_MODULE_PATH this project has
[... 215 lines cut ...]
Now run 'cmake --build /soft/freecad-source' to build FreeCAD
=================================================

-- Configuring done (12.0s)
CMake Warning (dev) in src/Base/CMakeLists.txt:
  Policy CMP0100 is not set: Let AUTOMOC and AUTOUIC process .hh files.  Run
  "cmake --help-policy CMP0100" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  For compatibility, CMake is excluding the header file(s):

    "/soft/freecad-source/src/Base/swigpyrun.hh"

  from processing by AUTOMOC and AUTOUIC.  If any of the files should be
  processed, set CMP0100 to NEW.  If any of the files should not be
  processed, explicitly exclude them by setting the source file property
  SKIP_AUTOGEN:

    set_property(SOURCE file.hh PROPERTY SKIP_AUTOGEN ON)

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done (2.6s)
-- Build files have been written to: /soft/freecad-source
```

hasecilu, 2023-11-18:
I tried to reproduce on latest but got a successful build.
Are you running `cmake` before? Are you using a different dir for compiling?
```shell
source_dir="$HOME/git/github/FreeCAD"
build_dir="$HOME/git/github/FreeCAD_build_0.22"

cmake -S $source_dir -B $build_dir    # 20[s]
cmake --build $build_dir --parallel 8 # First time 45[min], next 10[min]

cd $build_dir
time make -j$(nproc --ignore=2) # 10[s]
```

superbonaci, 2023-11-18:
Yes my dir is different than yours, but that shouldn't matter ?

superbonaci, 2023-11-18:
```
$ cd /git

$ git clone https://github.com/FreeCAD/FreeCAD.git freecad-source

$ cd freecad-source/

$ git submodule update --init
Submodule 'src/3rdParty/OndselSolver' (https://github.com/Ondsel-Development/OndselSolver.git) registered for path 'src/3rdParty/OndselSolver'
Cloning into '/git/freecad-source/src/3rdParty/OndselSolver'...
Submodule path 'src/3rdParty/OndselSolver': checked out '84762ac74ec6fe73a710f91ba3697a5034221537'

$ cmake . -DPYTHON_EXECUTABLE=/usr/bin/python3
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiler: GNU, version: 13.2.1
-- Looking for GL/gl.h
-- Looking for GL/gl.h - found
-- Looking for C++ include istream
-- Looking for C++ include istream - found
-- Looking for C++ include ostream
-- Looking for C++ include ostream - found
-- Looking for C++ include fstream
-- Looking for C++ include fstream - found
-- Looking for C++ include sstream
-- Looking for C++ include sstream - found
-- Looking for C++ include ios
-- Looking for C++ include ios - found
-- Looking for C++ include iostream
-- Looking for C++ include iostream - found
-- Looking for C++ include iomanip
-- Looking for C++ include iomanip - found
-- Looking for C++ include iostream
-- Looking for C++ include iostream - found
-- Check for STD namespace
-- Check for STD namespace - found
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.27.8
-- Compiling with Qt 5
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
=====================================================
Doxygen not found, will not build documentation.     
=====================================================

-- Checking for connection to GitHub...
-- GitHub connection established for FetchContent
-- Module support is disabled.
-- Version: 9.1.0
-- Build type: 
-- CXX_STANDARD: 17
-- Performing Test has_std_17_flag
-- Performing Test has_std_17_flag - Success
-- Performing Test has_std_1z_flag
-- Performing Test has_std_1z_flag - Success
-- Required features: cxx_variadic_templates
[... 538 lines cut ...]
Now run 'cmake --build /git/freecad-source' to build FreeCAD
=================================================

-- Configuring done (39.7s)
CMake Warning (dev) in src/Base/CMakeLists.txt:
  Policy CMP0100 is not set: Let AUTOMOC and AUTOUIC process .hh files.  Run
  "cmake --help-policy CMP0100" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  For compatibility, CMake is excluding the header file(s):

    "/git/freecad-source/src/Base/swigpyrun.hh"

  from processing by AUTOMOC and AUTOUIC.  If any of the files should be
  processed, set CMP0100 to NEW.  If any of the files should not be
  processed, explicitly exclude them by setting the source file property
  SKIP_AUTOGEN:

    set_property(SOURCE file.hh PROPERTY SKIP_AUTOGEN ON)

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done (1.2s)
-- Build files have been written to: /git/freecad-source
```

hasecilu, 2023-11-18:
> Yes my dir is different than yours, but that shouldn't matter ?

I meant if you're using a separate directory to build, you can build in the same directory where you have the source but it's better to use another dir to build. I had problems when I use the same dir.

superbonaci, 2023-11-18:
> I meant if you're using a separate directory to build, you can build in the same directory where you have the source but it's better to use another dir to build. I had problems when I use the same dir.

yes in-source building.

superbonaci, 2023-11-18:
I think the readme does not explain correctly how to build with out-of-source build:
https://wiki.freecad.org/Compile_on_Linux

hasecilu, 2023-11-18:
Yes, is not very clear, I figured out the command outside the wiki, I could edit the wiki but give some minutes.

superbonaci, 2023-11-18:
> Yes, is not very clear, I figured out the command outside the wiki, I could edit the wiki but give some minutes.

what is the command?

hasecilu, 2023-11-18:
When cloning for first time instead of using:
```shell
git clone https://github.com/FreeCAD/FreeCAD.git freecad-source
git submodule update --init
```
it's shorter to use:
```shell
git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD.git freecad-source
```

to avoid error when: `cmake -S $source_dir -B $build_dir`/`cmake ../freecad-source`

xtemp09, 2023-11-18:
I've had recently the same problem. Solved in similar way. [This commit](https://github.com/FreeCAD/FreeCAD/commit/94f7398f361fa1ffd91ef87c672e6561c787e436) introduces it.

It raises another question: if FreeCAD now uses submodules, would it be better to add other libraries as well, like _zipios_, _libfmt_ in a form of submodules?

hasecilu, 2023-11-18:
> what is the command?

It's the snippet I previously posted here and in the other bug report.

I have updated the wiki to have `--recurse-submodules` flag on git clone

> It raises another question: if FreeCAD now uses submodules, would it be better to add other libraries as well, like _zipios_, _libfmt_, _Boost.Signals2_ etc as submodules?

I think it's a good idea

Pesc0, 2023-11-21:
I managed to build just fine on archlinux, is this still relevant?

***

> It raises another question: if FreeCAD now uses submodules, would it be better to add other libraries as well, like zipios, libfmt, Boost.Signals2 etc as submodules?

CC @adrianinsaval

adrianinsaval, 2023-11-21:
I would expect this is related to https://forum.freecad.org/viewtopic.php?p=719786#p719786 rather than anything to do with ondselsolver

Pesc0, 2023-11-21:
The ping was just for the submodules subject ;)

adrianinsaval, 2023-11-21:
I too believe that we should use submodules for 3rdParty code but I don't know where the rest of the team stands on this subject

adrianinsaval, 2023-11-21:
I'll assume this is a duplicate of #11449 since I know for a fact that ondselsolver compiles on archlinux and the message that was posted in the OP is of a `warning` related to ondselsolver, not an error

closed 2023-11-21 by adrianinsaval