FreeCAD/FreeCAD #11439: [Problem] out-of-source build doesn't even start: make: *** No targets specified and no makefile found.  Stop.

superbonaci, 2023-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

out-fo-source building:
```
$ mkdir freecad-build
$ cd freecad-build
cmake ../freecad-source
...
-- Generating done (1.3s)
-- Build files have been written to: /git/freecad-source
$ make -j$(nproc --ignore=2)
make: *** No targets specified and no makefile found.  Stop.
```
according to
https://wiki.freecad.org/Compile_on_Linux

### Full version info

```shell
no info
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-18:
Please reference the exact place in the wiki page you're referring to

superbonaci, 2023-11-18:
https://wiki.freecad.org/Compile_on_Linux#:~:text=the%20best%20option.-,Out%2Dof%2Dsource%20build,-Building%20in%20a

hasecilu, 2023-11-18:
As I said before I had problems running the "bare cmake" command from the wiki and used another commands. The build dir is not a subdirectory of source dir.

Did you tried the snippet? Try using the cmake flags:
```shell
source_dir="$HOME/git/github/FreeCAD"
build_dir="$HOME/git/github/FreeCAD_build_0.22"
# copies code from source dir to build dir
cmake -S $source_dir -B $build_dir
# compiles on the build dir
cmake --build $build_dir --parallel 8
# change dir and make binary file
cd $build_dir
time make -j$(nproc --ignore=2) # 10[s]
```

hasecilu, 2023-11-19:
> make: *** No targets specified and no makefile found.  Stop.

This message appears when you execute `make` on a directory that hasn't build the code first, probably in the source dir instead of the build dir.

Did you already successfully build FreeCAD?

VM4Dim, 2023-11-19:
```
$ mkdir freecad-build
$ cd freecad-build
cmake ../freecad-source
...
-- Build files have been written to: /git/freecad-source
```
Probably CMake already was running in `/git/freecad-source` directory and CMakeCache.txt exist there.

https://cmake.org/cmake/help/latest/manual/cmake.1.html

`cmake [<options>] <path-to-source>`

 Uses the current working directory as the build tree, and <path-to-source> as the source tree.  The source tree must contain a CMakeLists.txt file and must not contain a CMakeCache.txt file because the latter identifies an existing build tree.

`cmake [<options>] <path-to-existing-build>`

   Uses <path-to-existing-build> as the build tree, and loads the path to the source tree from its CMakeCache.txt file, which must have already been generated by a previous run of CMake.

superbonaci, 2023-11-19:
This works now:
```
$ cd /git
$ git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD.git freecad-source
$ mkdir freecad-build
$ cmake -S freecad-source -B freecad-build
$ cd freecad-build
$ make -j$(nproc --ignore=2)
```
So the compile guide does not state well how to compile out-of-source:
https://wiki.freecad.org/Compile_on_Linux#:~:text=cd%20freecad%2Dbuild-,cmake%20../freecad%2Dsource,-make%20%2Dj%24(

the wiki has to be updated.

> Did you already successfully build FreeCAD?

@hasecilu  no still not #11438

flaviut, 2023-11-23:
The code [at Out-of-source Build](https://wiki.freecad.org/Compile_on_Linux#Out-of-source_build) works fine for me on latest Arch Linux. From my experience with CMake, it should also work fine everywhere. That's not to say your suggestion is wrong, it should do the same thing.

> ```
> $ mkdir freecad-build
> $ cd freecad-build
> cmake ../freecad-source
> ...
> -- Generating done (1.3s)
> -- Build files have been written to: /git/freecad-source
> $ make -j$(nproc --ignore=2)
> make: *** No targets specified and no makefile found.  Stop.
> ```

Did you try deleting the `freecad-build` folder and retrying?

---

That said, I'm not really happy with how this wiki page is written; there are some inaccuracies about what an out of source build is and if anything else should be used (in-source builds should never be used). I've chosen to [Be Bold](https://en.wikipedia.org/wiki/Wikipedia:Be_bold) (hopefully that's the policy here as well) and improve it a bit.

superbonaci, 2023-11-23:
@flaviut thanks for fixing the wiki. Now working so closing this.

closed 2023-11-23 by superbonaci