FreeCAD/FreeCAD #11441: [Problem] Combining STEP files messes tree structure in resulting file

superbonaci, 2023-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm using FreeCAD to combine 2 STEP files into one, like if it was originally one STEP file.
Each step file has a hardware piece, made each of 2 Mesh with one polygon inside each mesh.

I open the 2 step files at the same time with FreeCAD, then I drag the second next to the first, under the same `Unnamed`, and export as STEP. So my idea is to "combine" them like if the two pieces came in the same step.

The issue is that the structure/geometry tree, is not the same anymore. For example if you import the resulting step with Autodesk Navisworks Manage/Simulate, the tree is way scrambled.

Here are the FreeCAD screenshots of how I made the resulting STEP:

[image]

[image]

[image]

And this the Navisworks Manage tree importing the 2 steps independently and the combined one:

[image]

[image]

The 2 original files and the combined one:
[FreeCAD_step_files.zip](https://github.com/FreeCAD/FreeCAD/files/13401179/FreeCAD_step_files.zip)

### Full version info

```shell
FreeCAD 0.21.1 for Windows.
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

superbonaci, 2023-11-18:
This is how I want it to look like:

[image]

How must the resulting step file look like to do this? Any documentation? Can I just concatenate the 2 step files or does it involve complicated steps?

superbonaci, 2023-11-28:
@luzpaz in what aspect do you exactly need help? To know exactly how the resulting file must look like?

luzpaz, 2023-11-28:
Help fixing the issue :wink:

superbonaci, 2025-01-29:
Do not stale.

ElementW, 2026-09-14:
This issue or the framing of it is bogus.

First, this is most definitely not a *bug*: there is no noticeable malfunction either within FreeCAD itself or with the STEP file that is exported; you can import the merged file and the geometry & features are correctly presented *and processed* by Navisworks, meshers, CAD packages, etc.

Second, @superbonaci
> To know exactly how the resulting file must look like?

yes. It is unclear what's the list of things you consider wrong here: is it the structure is not exactly the same as the input? Is it the extra empty geometric sets (or compounds as called by [Mayo](https://github.com/fougue/mayo)), is it the nesting of meshes inside a bigger one? Is it the polygons' primitives or their declaration order? All of the above?

If it's reproducing the *exact* structure for any given input STEP set, I don't think that will ever happen. Importing converts geometry to OCC's internal representation whose structure when re-exporting to STEP is not guaranteed to re-create the original, and my bet is that's not planned (maybe @pieterhijma can pitch in on that).
Beyond that, it's not FreeCAD's prerogative, it's a CAD package not a STEP toolkit. There's other software out there for that, OCC-based or not.

As for the useless geometric sets, sure that could be improved, but the structural cleanliness of the STEP output is not guaranteed by FC. Which is why I'd reclassify this as a Task, or as a Feature if you break it down into child Task items. Now that's provided it's even actionable by us, which it probably isn't; [there's only so many settings we can tweak](https://occt3d.com/dev/doc/overview/html/occt_user_guides__step.html).